artisan-roaster-scope/artisan #1257: Artisan spontaneously disconnects from HotTop KN-8828B-2K+ with MacBook Air M2

avimentor, 2023-08-24:
Artisan will spontaneously and unexpectedly disconnect from the Hottop roaster, which creates a fair amount of excitement. I usually have to dump the beans by pulling and holding the manual drop knob on the Hottop. Depending on the stage of the roast, I may end up throwing out the beans and starting from scratch

Reproducing this issue is hard because it happens randomly. I've tried the following, but am not sure what allows Artisan to reconnect to the Hottop. Not sure if other Mac users are seeing this. I sometimes do three separate roasts in a row without issue. Other times, the spontaneous disconnect happens repeatedly.

To recover:
1. Unplug the Hottop, Quit Artisan, and launch again
2. If the disconnect happens again, I've tried rebooting the MacBook, then launching and connecting Artisan again, which seems to help. Sometimes, just quitting Artisan and re-launching is sufficient.

Perhaps this is a serial interface issue due to Artisan running under Rosetta emulation? Might a Apple silicon native version of Artisan help, when it becomes available?

Artisan 2.8.4
MacOS 13.5.1
MacBook Air M2
Connected to Hottop KN-8828B-2K+ via USB cable and an Apple-branded USB-A to USB-C adapter.

MAKOMO, 2023-08-25:
We are aware that there are issues with Artisan on M1/M2 macs. We still do not have a clear picture of the causes. The loose of connectivity to the Hottop is new to us. Thanks for reporting.

For the lack of an M1/M2 mac we can currently not investigate this much further. However, we have published a test build that might work more stable on M1/M2 macs. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/m1-test

Could you please test if this one behaves any better for you?

avimentor, 2023-08-25:
Hi Marko,

Thank you for your quick response.

I installed the M1-test version, but when I launched the app it just endlessly bounced in the task bar. I ended up force quitting it.

Can the M1 version co-exist with the legacy Artisan 2.8.4 version or should I remove the old version before launching the M1 test?

Thanks,

MAKOMO, 2023-08-25:
When you launch a new version it takes about twice as long to start as normally, especially if you start it from the image it got delivered. Just copy it over to your internal disk in some location and rename it into "Artisan Test" or something. It can co-exists with any other Artisan app on your disk. It uses the very same Artisan settings too.

avimentor, 2023-08-26:
I downloaded the M1-test version, mounted the image, and copied the app to my applications folder as ArtisanTEST.

I launched ArtisanTEST and waited for it to finish initializing. It did take a while to complete loading.

Used the same USB cable with an Apple-branded USB-A to USB-C adapter.

Clicked on the ON button.

Right-clicked on the timer to enable "super on".

Clicked on CONTROL button and the upper left display shows "Hottop control turned on" but the Hottop does not wake up and become active.

If I quit and launch Artisan 2.8.4 and do the same sequence, the Hottop wakes up and I can control.

Under Config->Device->Device assignment, I noticed a logging option at the bottom. Does turning this on log any useful data? If so, where is that data stored and perhaps I can post the contents of the log for troubleshooting?

MAKOMO, 2023-09-20:
> Under Config->Device->Device assignment, I noticed a logging option at the bottom. Does turning this on log any useful data? If so, where is that data stored and perhaps I can post the contents of the log for troubleshooting?

I just added device logging for the Hottop and made a new [M1 Test 2](https://github.com/artisan-roaster-scope/artisan/releases/tag/m1-test2) test release.

Turn device logging on (menu Config >> Device >> Device assignment) and best also the debug logging (hold the OPTION and COMMAND keys and klick on the plus icon in the upper left corner to toggle debug logging). The logs can be accessed by a click on the plus icon while holding just the OPTION key. Please send along those log files also your artisan-settings.aset (export via menu Help >> Export Settings).

Communication problems with the Hottop were reported before by (non M1) mac users for older versions and mostly were due to a broken cable, broken Hottop electronic or the installation of the FTDI serial driver. The last point happens over and over again. There is no need to install an FTDI driver on any of the modern OS versions as it comes with the system and installing another one can lead to communication issues. So if you installed an additional FTDI driver, uninstall it before testing.

Thanks for your support!

avimentor, 2023-09-25:
Installed the M1 Test 2 and tried running it. Just sent the log files and the current configuration for my MacBook Air.

Still unable to control the Hottop or display any temperatures.

MAKOMO, 2023-09-29:
John, thanks for spending your time test this build and send in your logs. It turned out that adding those log statements made the Hottop connection fail completely for some technical reasons. The multiprocessing used to implement the Hottop connection is not directly compatible with the logging infrastructure Artisan is using. I should have checked this before. That multiprocessing approach is most likely also the reason for the issues you experience under Apple Silicon and dates back to 2015.

I spend my day to reimplement the Hottop communication module completely using a modern asyncio approach. I tested this using my Hottop, but I don't have access to an Apple Silicon machine to verify that it works on M1/M2 machines.

Here is a signed and notarized build with this new Hottop implementation.

https://github.com/artisan-roaster-scope/artisan/releases/tag/m1-test3

I would be happy if you could test if this one does any better and report back. This one now logs correctly any communication between Artisan and the Hottop if the device logging is turned on in the Device dialog. As there are also some smaller changes in the Hottop setup I suggest to run the Hottop machine setup again after installing this build.

avimentor, 2023-09-30:
Marko,

Thank you for the new build. I launched it, reconfigured the Hottop device, and was able to connect.

I was verifying that all the controls worked (temp, fan, door open/close, stirrer on/off, drop), and the Hottop spontaneously disconnected. Was able to reconnect by quitting Artisan and relaunching.

I've emailed the log files, but was when I tried to save the settings Artisan hung with the spinning beachball.

So progress is definitely being made!

avimentor, 2023-10-09:
Marko,

The latest M1 build works great on my 2002 MacBook M2. Thank you!

I've verified the spontaneous disconnect was being caused by electro-magnetic interference from the dual motor VentAHood exhaust fan operating over the Hottop roaster: Turning either of the two VentAHood fans on or off reliably causes Artisan to disconnect from the MacBook using the 5 foot USB cable supplied with the roaster connected through an Apple-branded USB A-to-C adapter.

I followed your suggestion and replaced the 6-foot USB-A cable with a 1-foot USB-A cable.

https://www.amazon.com/dp/B0C7GDM3Y4?ref=ppx_yo2ov_dt_b_product_details

Also as you suggested, I ordered some snap-on ferrite coil noise suppression filters. Since they came as a pack of 10, I installed five on the USB cable and one on the Apple-branded USB-A-to-C adapter. That might be overkill, but I had 10 available so I figured what the heck.

https://www.amazon.com/dp/B08DC9JF5K?psc=1&ref=ppx_yo2ov_dt_b_product_details

With this set-up, I found that Artisan M1/M2 version 5 running on my MacBook M2 remained reliably connected to the Hottop regardless of how many times I cycled the VentAHood fans on or off. Problem solved: No more spontaneous disconnects!

[image]

[image]

closed 2023-10-09 by avimentor