artisan-roaster-scope/artisan #1267: Button can trigger an alarm

Robjones6146, 2023-09-28:
**Feature request:**
A Button triggers an alarm. I want to trigger an alarm (Nr.1) when I click a button so that I can then make another If alarm 1 to trigger x seconds later.

**What for?**
My example:
The button in this case is a multiple Event button called **LOAD** that 1) switches on the green bean loader blower, 2) opens the bean inlet slide to the roast chamber, 3) opens the slide of the green bean loader hopper. The second alarm then triggers an Event Button x seconds later that switches off the blower and closes the slide actuators. Now I can get an alarm Pop Up to remind the roaster to increase the Loft (Blower) to 15kg (for a fluid bed roaster) after the beans are loaded. Note: The roast machine is on another floor.

**Another example:**
I click a Button **HEAT UP** that's unrelated to a time/temperature Event, eg. FC, which then activates an alarm. Now the user can create a chain of alarms based on that one click, say, increase the heat 5 degrees after 5 seconds.

**Motivation**
As more commercial roasters use Artisan, this will be a valuable feature being able to control external equipment, pre and post-roast - directly related to the roast process.

Maybe I have missed a way to trigger an Event x seconds after the click on a Button. Please advise.

I know you are busy. I see there are lots of good/useful other feature requests and you need to select/decline/balance feature requests.

**How would it work?**
One possibility is to use the From tab. From >> BUTTON >> No. (would require another Tab/field - Button No.). Just an idea.

Thank you for all the good work Marco, and others. I know you have had this request before, but there was a better existing solution for that request.

MAKOMO, 2023-09-28:
> **Feature request:** A Button triggers an alarm. I want to trigger an alarm (Nr.1) when I click a button so that I can then make another If alarm 1 to trigger x seconds later.
>
> **What for?** My [...]

Why do you need alarms for this at all? Just make use of the `Artisan Command` `sleep(n)` in a sequence of button actions (separated by semicolons) as for example here.

[image]

> **Another example:** I click a Button **HEAT UP** that's unrelated to a time/temperature Event, eg. FC, which then activates an alarm. Now the user can create a chain of alarms based on that one [...]

Artisan is mostly used by commercial roasters;)

Why not use different alarm sets, one for heat up, one for roasting and switch between those via buttons/alarms using the `Artisan Command` `alarmset(<as>)`?

> Maybe I have missed a way to trigger an Event x seconds after the click on a Button. Please advise.
>
> I know you are busy. I see there are lots of good/useful other feature requests and you need to [...]

closed 2023-09-28 by MAKOMO

Robjones6146, 2023-09-28:
Hi Marco,

Thanks, that works for a message. But how do I trigger an Event? I want to trigger a button 30 seconds after the first button.

[image]

I want to click the START button to open the slide, and then activate the STOP button 30 seconds later.

Robjones6146, 2023-09-28:
That works for a message, not for an Event.

How do I trigger another Event 30 seconds after the first Event?

MAKOMO, 2023-09-28:
You can use a sleep(30) command in a Multiple-Event action list (I extended the documentation accordingly), but current Artisan does not allow to call a Multiple-Event action from a Multiple-Event action to prevent non-terminating loops.

I just lifted this restriction which allows you to do something like the following.

[image]

A new continuous build containing this (kind of dangerous) change is available here:
https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Robjones6146, 2023-09-28:
Thanks marko,

That is what I tried earlier. Will try again now after download (see
attached)

I have an Alarm for START that ends all three events (Off, close, close).
Will that terminate any possible loop?

Robjones6146, 2023-09-28:
My example below works. Thank you so much!!

[image]

When I press START it activates the three "off" Events (grouped under Event 2 above) - Will this work as a failsafe to prevent any non-terminating loops?

[image]

As a matter of interest, what is "sleeping" as other Artisan Events and Alarms continue to work... while the selected event sleeps?

MAKOMO, 2023-09-28:
I don't understand your button action 45. This one does nothing but sleeping and is OFF. What is this for?

Non-termination can occur if you activate a button from another's button action and that button itself again (maybe via some chain of button actions or directly) fires again that other button back.

Sleep is non-blocking.

Robjones6146, 2023-09-28:
Event No. 45  is an  Artisan Command -- sleep(30) --- it works

Robjones6146, 2023-09-29:
One final question on this topic. How can I get this "Multiple-Event action from a Multiple-Event action to prevent non-terminating loops" included if I download a later version next year...say Artisan 2.5.7

MAKOMO, 2023-09-29:
I don't understand your question. You simply should avoid to specify a circular call loop with your button definitions. That is on you now. In previous versions you simply couldn't call a multi-event button action from a multi-event button. Now you can due to your request.

Robjones6146, 2023-09-29:
You said:  *I just lifted this restriction which allows you to do something
like the following....*

I wanted to know if this is now a permanent feature or just for this beta
build. Are you saying this is now a permanent feature going forward in new
releases?

Please can you give an example of a circular call loop.  Do you mean 1
calls 2 and then 2 calls 1 ?

MAKOMO, 2023-09-29:
> You said: *I just lifted this restriction which allows you to do something like the following....* I wanted to know if this is now a permanent feature or just for this beta build. Are you saying this is now a permanent feature going forward in new releases? 

Permanent

> Please can you give an example of a circular call loop. Do you mean 1 calls 2 and then 2 calls 1 ?

Yes