artisan-roaster-scope/artisan #1268: On clicking Event Button... Are you sure?  YES - CANCEL

Robjones6146, 2023-09-29:
**Not strictly a feature request. It would be very useful to avoid a disaster but not very often.**

 **My issue:**
I have a button **Burner off** 
If this button is clicked by mistake, the roast will be ruined as it takes 37 seconds to re-ignite the Reillo burner (5-minute roast time, fluid bed roaster) I also have a kind of emergency **SHUT DOWN** button that turns everything off and closes everything.

**Solution:**
Config >> Events >> Config >> Event Pop Up qualify >> YES/CANCEL  (with checkbox)

Marko, I am sure you already have an elegant solution.
[image]

MAKOMO, 2023-09-29:
This project accepts PRs. We are looking forward for your contribution!

Robjones6146, 2023-09-29:
Thank you.

On a separate issue. The roaster is currently on a ship destined for the
UK. I have requested the owner to make a donation once the roaster is up
and running.

maxoliverbr, 2023-11-20:
@MAKOMO Hi ! I was checking how this could be done, maybe if we create a event action called "confirm()":
`confirm();button(True); `

That would trigger a confirmation dialog and if "No" is clicked would exit the event. 

Or change the behavoir of the popup() action to add a Yes/No option.

Robjones6146, 2023-11-20:
Marko would know best.

I really like the popup idea. The message is already there, add OK/CANCEL option. Works with Artisan Command, Multiple Event, and possibly in an Alarm sequence. The first Event is the alarm that pauses or suspends the second Event until the user Accepts or Cancels the second Event in the sequence.

popup(<msg>[,<int>],[<opt>])

popup(Are you sure?, 0, 1) 

Would be great if you could do this Max.

MAKOMO, 2023-11-20:
Not trivial to impossible. Event actions are evaluated in a parallel thread which is guarded by semaphores. All GUI actions like popups need to be evaluated in the GUI thread. Blocking threads waiting for user actions is problematic. The current popup action is not connecting back to the event thread thus is not problematic

closed 2023-11-20 by MAKOMO