artisan-roaster-scope/artisan #1272: Hi! I sometimes have issues sometimes when noting/assigning the time First Crack. I love that there is the option to retroactively add First Crack if you've missed it initially, but when you right click to reach the option to add FC the option to add/reassign Charge also exists. On several occasions I have accidentally clicked Charge (I work on a touch screen device) and it's shifted the entire roast profile over to the point where I'm just confused as to where I am. Is it possible to move the "reassign Charge" feature to somewhere else, like the bottom of the screen, and simply allow the reassigning of FC?

christinamariehughes, 2023-10-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bigl00z3, 2023-10-01:
Erm.... first this is not a bug or an issue.
and 2nd, if you look at the template, you are supposed to fill the form if you want others to be able to help you.
Filling the full text in the topic does not help and will not motivate others to support your request.
Do you also write your full email in the subject line?

Regarding your first crack question. You can also if you think you need to move it later, you can always cancel the marking with the keyboard arrows (<-) followed by space bar and then reassign a bit later. Being able to edit via right click (including for the charge tag) is a great function and is very useful.

Nico

christinamariehughes, 2023-10-01:
Alright, sorry. It’s my first time writing in about anything so I apologize
for not jumping through the right hoops.

MAKOMO, 2023-11-18:
I understand that one could accidentally move CHARGE instead of FCs by using this right-click context menu, but there is the case you actually want to move CHARGE. Moving entries around will not avoid this confusion. A user is always free to "press the wrong button". I personally rather re-set the CHARGE and never the FCs event.

Thanks for raising this issue. We will keep it in mind until we find a solution that servers everybody, but close this issue for now.

closed 2023-11-18 by MAKOMO