artisan-roaster-scope/artisan #1288: New Rpi version "bookworm" issue loading background profiles using button withArtisan command  loadBackground("...")

bigl00z3, 2023-10-27:
**Describe the bug**
I installed the "RPi Test version" 2.8.5 (f345c5f7). The artisan buttons I have to quickly load background profiles are giving the error "file not found"
The same button was working on previous Rpi version. Newly created buttons (with checked path) are giving the same error. The files are in the same location/path as before

**To Reproduce**
Steps to reproduce the behavior:
1. Edit the path of the required button to an existing alog file (using the command `loadBackground("/home/pi/myfolder/mybgprofile.alog"`  )
2. Press button to load profile
3. Error message "file not found" with the path of the file
4. The error is happening with existing buttons (from previous setup) but also with newly created buttons (getting the new path in case the new install had changed something in the path)

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:  2.8.5 
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: (f345c5f7)
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Raspberry Pi Bookworm (on a Rpi 4b)
 - Connected devices or roasting machine [e.g. Probatone 5]: Phidgets and modbus
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/13192103/artisan-settings.txt)

bigl00z3, 2023-10-28:
It seems that it is due to the fact that on the new install (either due to bookworm or artisan) the undescores "_" in the path are not recognized. In the error message I see them as "0".
It used to work in the previous install.
Nico

MAKOMO, 2023-10-29:
Confirmed. This is a regression over Artisan v2.8.2. Thanks for reporting!

MAKOMO, 2023-10-29:
Fixed by Commit 5ab340d04804f79f1170c01fa447480d027f45fd. New build uploaded at the same location

closed 2023-10-29 by MAKOMO

bigl00z3, 2023-10-29:
Thanks, it works. After a slightly bumpy install.
Just for the note:
I tried to install the new versions as I was doing it before (right click on the package and select "Package Install") but I still had the issue when I started artisan. After checking I still had the same build number. 
So I tried installing from the command line and it was in fact not installing as it is the same version. I got:

```
pi@pi:~/Desktop $ sudo apt install artisan /home/pi/Desktop/artisan-linux-2.8.5_raspbian-bookworm.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'artisan:armhf' instead of '/home/pi/Desktop/artisan-linux-2.8.5_raspbian-bookworm.deb'
artisan:armhf is already the newest version (2.8.5).
artisan:armhf is already the newest version (2.8.5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

So I removed it with `sudo apt remove artisan`
and reinstalled it normally. Now it's all good and the fix is working.

Nico

MAKOMO, 2023-10-30:
Sorry Nico. I assume this happend as I did not modify the version number of this package. The install system assumed that this Artsian package is already installed (same version number 2.8.5) and thus did not install it again. Installing with the "-f" option for a forced install should work in this case. The install with the right-click will obviously never succeed if the version number of the installed package is the same as one to be installed. Thanks for the hint. I will add a minor build number to future beta builds.