artisan-roaster-scope/artisan #1293: Hottop sets fan to 20% instead of 20%

RickGroszkiewicz, 2023-11-13:
I am upgrading my roasting laptop to the latest version of linux Mint, and the latest version of artisan. Based on past experience, I did not try to make changes to my prior setup. I wiped the hard drive and installed both linux Mint 21.2 and artisan 2.8.4 from scratch.

I have encountered a problem where my alarm file sets the fan to 25%, but artisan sets the fan to 20%. Then I have to manually change the fan to 25%. I have attached screen shots of the roast events and the roast alarms. 

[image]

[image]

I had a similar problem a few years ago, using a prior version of artisan. The solution was to NOT use the "coarse" option under Roast / Events / Quantifiers:
http://www.softwarepolish.com/personal/Artisan-Reinstalled-Linux-Mint-B2K/album/index.html#img=B-2K%20change%20quantifiers%20to%20match%20event%20types.png
I have not been able to locate that "coarse" option using artisan 2.8.4.

MAKOMO, 2023-11-13:
Rick, the Hottop allows to set the fan speed (Air) only in 10 steps. Thus the Artisan sliders range 0-100 is divided by 10 and should be set to steps of 10 instead of 1 as the default. Note that this "Coarse" setting was replaced by the step size parameter which can be 1, 5 or 10. You want the step size of "Air" for the Hottop to be 10 for the slider as well as the quantifier.

[image]

[image]

An alarm action to set the Hottops fan to 25 does not much sense as this will be "quantified" to the next valid step which is 20.

I am closing this as this is not a bug, but a configuration issue.

closed 2023-11-13 by MAKOMO

RickGroszkiewicz, 2023-11-13:
So I guess this changed since artisan 2.4
All my alarm files for the Hottop always use heat of 90% and fan of 25% at Dry End.