artisan-roaster-scope/artisan #1295: No longer able to download previous release builds from GitHub

tleef, 2023-11-15:
**Is your feature request related to a problem? Please describe.**
In my case v2.8.4 is crashing on my M1 Macbook Air. I'd like to downgrade to v2.8.2 but can't find the .dmg
I think this problem can be generalized to more then my specific scenario however.

**Describe the solution you'd like**
Make sure that previous release builds are able to be downloaded from the GitHub releases page.

**Describe alternatives you've considered**
Host them somewhere else

**Additional context**
This should be the link to the v2.8.2 build for Mac but at the time of this write it gives a 404
https://github.com/artisan-roaster-scope/artisan/releases/download/v2.8.2/artisan-mac-2.8.2.dmg
Add any other context or screenshots about the feature request here.

MAKOMO, 2023-11-18:
We don't offer older versions for download nor do we provide support for older versions due to limited resource we have to run this project at all. We hope for your understanding and support.

There is a test version of the next Artisan release version (to be released later this month) that runs native on M1 and is reported to not suffer from those crashes. You can find it here:

https://github.com/artisan-roaster-scope/artisan/releases/tag/m1-test-native

closed 2023-11-18 by MAKOMO

jonasskafte, 2024-06-18:
Hi there @tleef,

Did you find a solution to this? I'm in a bit of a pickle as I've just updated Artisan without checking compatibility with my old mac and I now cannot find a way to install the 2.8.2 version.

Have you, by any chance, got a copy of the DMG file that you can share with me? If so, I'll be you ever so grateful!

Thanks!

MAKOMO, 2024-06-18:
Older Artisan versions are not supported, nor available. Did you check the current version 2.10.4 which is build as universal binary and runs native on Intel as well as Apple Silicon (M1, M2, M3...)? It runs super smooth for many.

If you observe Artisan crashes we should investigate what goes wrong on your setup. For this we need some more information.

MAKOMO, 2024-06-18:
For older Mac or Windows machines not any longer supported by the current builds we offer legacy builds for both platforms in the very same download folder. Everything machine which was supported by Artisan v2.8.2 is still supported by the current release.

jonasskafte, 2024-06-18:
Thanks for your quick response @MAKOMO!

I've got an old MacBook Pro from 2012 with macOS Catalina installed. So the latest version of Artisan I am able to run is the v2.8.2 (newer versions simply won't open).

The problem is that I can only find the source code files in the [v2.8.2 folder](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.2) and I haven't found a way to install it from the source code files. [This guide](https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/Installation.md#step-1-download-artisan-for-your-platform) only refers to the DMG files and not the source code files.

I was therefore hoping that @tleef would still have a copy of the DMG file laying around, but if you can help me get back in action somehow, then I'd be super grateful!

MAKOMO, 2024-06-18:
> Thanks for your quick response @MAKOMO!
> 
> I've got an old MacBook Pro from 2012 with macOS Catalina installed. So the latest version of Artisan I am able to run is the v2.8.2 (newer versions simply won't open).

Why is this? The latest 2.10.4 macos legacy build supports macOS 10.13 and up and runs fine here on 10.15 Catalina.

Just download and install [artisan-mac-2.10.4-legacy.dmg](https://github.com/artisan-roaster-scope/artisan/releases/download/v2.10.4/artisan-mac-2.10.4-legacy.dmg).

> The problem is that I can only find the source code files in the [v2.8.2 folder](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.2) and I haven't found a way to install it from [...]