artisan-roaster-scope/artisan #1301: Option to hide Buttons Visibility with a Button Action

Robjones6146, 2023-11-17:
**Option to hide Buttons Visibility with a Button Action**

Is it possible to hide a button after it has been activated or stop it from being activated a second time during a roast?

I have tried activating another palette with the button visibility OFF, but could not get that to work.
I have requested an **Are you sure?**  YES/CANCEL  option Popup feature when a button is clicked (hopefully someone can add this)

**Reason:**
I have a lot of automation going on, Artisan also talks to outside software that does some stuff. Clicking my **Load Bean - START** button does 6 separate operations if this button is accidentally clicked during a roast this will result in loading the beans for the next roast. I also have other buttons that activate a sequence of events when clicked.

**Possible solutions:**
Config >> Alarm >> Action >> Deactivate Event
Config  >> Events >> Buttons >> Action >> Artisan Command >> visibilityOff(<Button number, true/false>)

Thank you in advance.

roasterdave, 2023-11-17:
`I have tried activating another palette with the button visibility OFF, but could not get that to work.`

I just tested this and it works properly in v2.8.4 and in the continuous build. 

[image]

Robjones6146, 2023-11-17:
I am on Artisan 2.8.5 (e969944)  but that should not make a difference

Thanks, how did you do that?

roasterdave, 2023-11-17:
`I am on Artisan 2.8.5 but that should not make a difference`
No difference.  As stated above I took the time to test both the continuous build (v2.8.5) and v2.8.4.

`Thanks, how did you do that?`
Two palettes each with three buttons.  Button number 2 has **Visibility OFF** in palette #2.  Button 3 uses the Artisan Command **palette(\<p\>)** to switch between palette #1 and palette #2.

[issue1301_palettes.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/13389318/issue1301_palettes.aset.txt)

Robjones6146, 2023-11-17:
That's what I am doing, but it doesn't save the on/off setting when I save
the palette

Robjones6146, 2023-11-17:
My palette on/off button disappears every time. It doesn't save when I save the Palette file

MAKOMO, 2023-11-17:
Daves example works as described. Also saving and loading the palettes to via an `.apal` file works. Note: the currently selected palette is not changed by loading palettes from an `.apal` file. The currently selected palette is part of the the system state and stored along the Artisan settings.

MAKOMO, 2023-11-17:
> I have requested an Are you sure? YES/CANCEL option Popup feature when a button is clicked (hopefully someone can add this)

Please don't mixup things and open a separate feature request. Note that to my assessment it is very very unlikely that someone implements this feature if not yourself. The project is open to contributions via PR which will be carefully reviewed and if productive on all platforms integrated.

Robjones6146, 2023-11-17:
Thanks, I see that his example works, but I can't get mine to work, I cant find any helpful documentation on this, I am sure I am doing something wrong, I need some help, thats all.

I created the 2 palettes, saved them, then saved the settings, then I clicked the newly created button, and it dissapeared, that happens every time, no matter what I do. I currently have two saved pallettes, saved to my settings file.

"_I have requested an Are you sure? YES/CANCEL option Popup feature when a button is clicked (hopefully someone can add this_" - I was mentioning a request I made a few weeks ago for the same problem I am facing.  https://github.com/artisan-roaster-scope/artisan/issues/1268

If I knew how to code, I would make it.

Robjones6146, 2023-11-17:
**Finally figured out this complicated process. For anyone with the same issue.**

In Events >> Buttons, create your buttons (on or off) including an (Artisan Command) for palette(1), lets call that button **default** palette
Now go to Palettes.
In current box write _default_, now select store #1 and then click Store Palette
Next you must save the palette

Now create the next palette
-- repeat the process above
close the Event dialog
Now you must save the settings in Help

Robjones6146, 2023-11-17:
Thank you all for your assistance.

closed 2023-11-17 by Robjones6146