artisan-roaster-scope/artisan #1323: i cannot see tp, fc, de on the profile after the roast

yarino25, 2023-11-27:
**Describe the bug**
**i cannot see tp, fc, de marks 
on the profile after the roast
only the charge in appearance** 

**Screenshots**
[image]

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.4
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: ac17fbe
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: mac
 - Connected devices or roasting machine [e.g. Probatone 5]: phidget 2xrtd

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[yarino 25.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/13478009/yarino.25.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2023-11-27:
Could you please also attach the profile shown in that linked screen shot (btw. you could have just drag-drop that screenshot here on a comment box to have it inline).

Please rename the *.alog file to end in *.txt or zip it before you drag-drop it here on GitHub to a comment box. A reply by email will not work!

yarino25, 2023-11-27:
[image]

yarino25, 2023-11-27:
[yarino 25.txt](https://github.com/artisan-roaster-scope/artisan/files/13480237/yarino.25.txt)

is that work?
thanks!!

yarino25, 2023-11-27:
[0030 גואטמלה שטוף.txt](https://github.com/artisan-roaster-scope/artisan/files/13480383/0030.txt)

MAKOMO, 2023-11-27:
Loaded your settings and your profile in Artisan v2.8.4 and it works for me. All annotations are visible.

[image]

yarino25, 2023-11-27:
so what could be the reason sone that it not visible ?

yarino25, 2023-11-27:
ok
i try to change the language from Hebrew to english and it works
so the problem is in the hebrew language

MAKOMO, 2023-11-27:
I can reproduce this now. Will try to fix it.... Thanks a lot for reporting!

Terracotta-6, 2023-11-28:
[image]

Terracotta-6, 2023-11-28:
[image]

MAKOMO, 2023-11-28:
What is this screenshot about?