artisan-roaster-scope/artisan #1334: 2.10.0 Hottop Control Button Not Working

bsperduto, 2023-12-05:
**Describe the bug**
After upgrading to 2.10.0 when I attempt to turn on the hottop control button nothing happens and the color remains the same. This resolved after downgrading to 2.8.4

**To Reproduce**
Steps to reproduce the behavior:
1. Configure a roast with a Hottop 2k+
2. Before turning on the roast right click on the time to turn on super
3. Click control button

**Expected behavior**
Roaster turns on and drum starts rotating

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.10.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows
 - Connected devices or roasting machine [e.g. Probatone 5]: Hottop 2k+

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2023-12-05:
I did check locally with my Hottop 2k+ with Artisan running on macOS and it worked as expected. I don't have a Windows machine to check, but the code is the same and I cannot see why it should not work there.

I assume you got readings from the Hottop, right?

Could you please share your Artisan log file? Hold down the ALT key and click the plus icon in the upper left corner of the Artisan main window. An email should open with the log files attached. Just send this email to info@artisan.plus. Thanks!

MAKOMO, 2023-12-05:
Now check on Windows 10 running in VirtualBox and it worked as expected.

bsperduto, 2023-12-06:
Sure, I sent a working log from 2.8.4 in case it's helpful and then the 2.10.0 logfile.

MAKOMO, 2023-12-06:
Artisan v2.10 is connecting to the machine only on ON and disconnecting on OFF. In previous Artisan versions one could click CONTROL (in super user mode) to get Artisan connected to the roaster and start controlling it. This is no longer possible. After pressing ON, Artisan v2.10 controls the Hottop as expected.

closed 2023-12-06 by MAKOMO