artisan-roaster-scope/artisan #1335: Simulator not working correctly in v2.10.0

Brewzologist, 2023-12-07:
**Describe the bug**
Upgraded to Artisan v2.10.0.  Ran Simulator and it worked fine.  Two days later did 5 roasts and it worked fine.  Today I ran Simulator and it only logs ET.  I downgraded to v2.8.4 and Simulator runs correctly with same .aset and .alog file.  Reinstalled v2.10.0 and problem remains.  Tried with multiple alog files and also my older aset files with no success.  I looked at the artisan.log file and it shows errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Load attached aset and alog files in Artisan v2.10.0
2. Run Simulator
3. Note that only ET is logging.  Also note errors in artisan.log

**Expected behavior**
Running Simulator should show all roast data from underlying alog file.

**Screenshots**
See attached.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: v2.10.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: c44a180
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10
 - Connected devices or roasting machine [e.g. Probatone 5]: Phidgets to Huky drum roaster

**Additional context**
None
[artisan-bug-stevek.zip](https://github.com/artisan-roaster-scope/artisan/files/13601951/artisan-bug-stevek.zip)

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

Brewzologist, 2023-12-07:
[artisan-bug-stevek.zip](https://github.com/artisan-roaster-scope/artisan/files/13602069/artisan-bug-stevek.zip)
Added example alog file that generates the error.  But note that any alog file generates this error

MAKOMO, 2023-12-07:
Indeed, the simulator of v2.10 struggles on profiles with extra device curves. I fixed this regression in Commit 0a0f80e6539d0d0fd1f12f005b157fa8308dd094.

A new continuous build incl. the fix should be online within about an hour at:
https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

MAKOMO, 2023-12-07:
Thanks for your excellent and detailed report that allowed me to fix this very quickly.

Brewzologist, 2023-12-07:
It's the least I can do. Thank you for all that you do for the coffee community!

Happy holidays;
Steve

closed 2023-12-13 by MAKOMO