artisan-roaster-scope/artisan #1346: Error log for Designer

NM12262, 2023-12-13:
Version 2.8.2            Revision : f525b16

<Class 'TypeError'> :
list indices must be integers or slices, not NoneType

In the "Designer" section, after selecting "config" and changing the numbers, this error occurred :

Error Log :
"
version = 2.8.2  2 12:51:27.538 Error: File "artisanlib\designer.py", line 384, in settimes -------------------------------------------------------------------------------- artisanlib\designer.py::settimes:384 self = _ = False strings = ['CHARGE', 'DRY END', 'FC START', 'FC END', 'SC START', 'SC END', 'DROP'] timecheck = 1000 checkvalue = 1000 timez = 776.03433652   1 12:51:16.072 Error: File "artisanlib\designer.py", line 606, in changeflags -------------------------------------------------------------------------------- artisanlib\designer.py::changeflags:606 self = _ = True sender = idi = 3 timez = 776.03433652 bt = 218.0 et = 225.0 newindex = None   

"

and this explanation :
"
File "artisanlib\main.py", line 16275, in desconfig 
File "artisanlib\designer.py", line 63, in __init__
File "artisanlib\designer.py", line 527, in loadconfigflags
  -------------------------------------------------------------------------------- 
artisanlib\main.py::desconfig:16275
	self = <artisanlib.main.tgraphcanvas object at 0x0000021D5C2086B0> 
		_ = False 	
	designerconfigDlg = <class 'artisanlib.designer.designerconfigDlg'> 
artisanlib\designer.py::__init__:63
	self = <artisanlib.designer.designerconfigDlg object at 0x0000021D7F6DE480>
	parent = <artisanlib.main.ApplicationWindow object at 0x0000021D5B35BB10>
	aw = <artisanlib.main.ApplicationWindow object at 0x0000021D5B35BB10> 
	charge = <PyQt6.QtWidgets.QLabel object at 0x0000021D78B43400> 
	drop = <PyQt6.QtWidgets.QLabel object at 0x0000021D78B41EA0> 
	__class__ = <class 'artisanlib.designer.designerconfigDlg'> artisanlib\designer.py::loadconfigflags:527 
	self = <artisanlib.designer.designerconfigDlg object at 0x0000021D7F6DE480>
"

And again, when I went to the "designer" section and selected config, after changing the "FC End" numbers, I checked the one I wanted to check, and this error was seen:
File "artisanlib\designer.py", line 606, in changeflags  
-------------------------------------------------------------------------------- 
artisanlib\designer.py::changeflags:606
	self = <artisanlib.designer.designerconfigDlg object at 0x0000021D78BD6480> 
	_ = True
	sender = <PyQt6.QtWidgets.QCheckBox object at 0x0000021D734CB370> 
	idi = 3
	timez = 776.03433652
	bt = 210.0
	et = 217.0 
	newindex = None 

and Error Log :
"
version = 2.8.2  1 13:40:04.476 Error: File "artisanlib\designer.py", line 606, in changeflags -------------------------------------------------------------------------------- artisanlib\designer.py::changeflags:606 self = _ = True sender = idi = 3 timez = 776.03433652 bt = 210.0 et = 217.0 newindex = None 
"

MAKOMO, 2023-12-13:
Duplicate: https://github.com/artisan-roaster-scope/artisan/issues/1173

This has been fixed for v2.8.4. Any reason why you are not using the current Artisan version which is v2.8.10?

closed 2023-12-13 by MAKOMO

NM12262, 2023-12-13:
Thanks,
It solved with artisan-win-2.10.0.