artisan-roaster-scope/artisan #1368: Artisan v2.10.0 can't load Yoctopuce drivers in Ubuntu

smaca69, 2023-12-29:
Artisan v2.10.0 can't load Yoctopuce drivers in Ubuntu.  

Steps to reproduce the behavior:
1. Connect Yoctopuce thermocouple via USB
2. Run Artisan and press ON button

**Expected behavior**
I get an error that flashes up and then disappears saying it can't load /usr/share/artisan/_internal/cdll/libyapi-amd64.so 

** Extra info **
I downloaded the latest file from Yoctopuce and copied it to the above location but I still get the error.
If I uninstall v2.10.0 and install v2.8.4 I dont get the error.

**Setup (please complete the following information):**
 - Artisan Version 2.10.0
 - Artisan Build (c44a180):
 - Platform (Ubuntu) 
 - Connected Yoctopuce Thermocouple

[artisan-settings-MILL_CITY.zip](https://github.com/artisan-roaster-scope/artisan/files/13791948/artisan-settings-MILL_CITY.zip)

MAKOMO, 2024-01-07:
Works fine out of the back on Ubuntu 22.04 for me. 

Both, Artisan v2.8.4 and v2.10.0 are build on Ubuntu 20.04. No difference on this side. Source code in regards to the Yoctopuce driver did not change in Artisan neither.

No idea on how to investigate on this one if I cannot duplicate it. Did you try to start Artisan v2.10.0 as super user? Does this issue persist after restarting of the machine?

MAKOMO, 2024-02-07:
Closing this as it seems to be stall.

closed 2024-02-07 by MAKOMO