artisan-roaster-scope/artisan #1376: IKAWA Home - MacBook Air M2 - OSX 14.2.1 - Bluetooth doesn't connect

drgarys, 2024-01-01:
Trying to startup. Bluetooth won't find the IKAWA Home as a device.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Bluetooth settings
Doesn't find IKAWA Home

The IKAWA Home roaster is on. The light near the Start button is alternating between red and green.

**Expected behavior**
I should see the IKAWA Home as a connected device.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Setup (please complete the following information):**
 - Artisan Version 2.10.0:
 - Artisan Build c44a1802
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.

The aset file is greyed out. But when I open Artisan in that configuration, it shows the aset file.

[image]

Please attach any relevant Artisan *.alog profiles.

I don't yet have alog files because I haven't been able to run Artisan for the IKAWA Home.

drgarys, 2024-01-01:
I found an additional setting that seems to be correct. 

Apple logo on upper left -> System Settings -> Privacy & Security -> Bluetooth -> Allow the applications to use Bluetooth

Artisan was checked. I unchecked it and cycled through restarting Artisan to reconnect it in Bluetooth. Bluetooth still doesn't find my IKAWA Home roaster.

drgarys, 2024-01-01:
If I open the IKAWA Home aset file in text edit, I get this code:

[General]
AUCLCDmode=0
AUCbase=212
AUCbaseFlag=false
AUCbegin=1
AUCguideFlag=false
AUClcdFlag=false
AUCshowFlag=false
AUCtarget=640
AUCtargetFlag=false
Alphas=@Variant(\0\0\0\b\0\0\0\x3\0\0\0\x18\0\x61\0n\0\x61\0l\0y\0s\0i\0s\0m\0\x61\0s\0k\0\0\0\x6?\xd9\x99\x99\x99\x99\x99\x9a\0\0\0\x10\0l\0\x65\0g\0\x65\0n\0\x64\0\x62\0g\0\0\0\x6?\xd9\x99\x99\x99\x99\x99\x9a\0\0\0$\0s\0t\0\x61\0t\0s\0\x61\0n\0\x61\0l\0y\0s\0i\0s\0\x62\0k\0g\0n\0\x64\0\0\0\x6?\xf0\0\0\0\0\0\0)
AmbientTempSource=0
AnalysisResultsLoc=0.49, 0.5
BTBColor=#0a5c90
BTBdeltaColor=#0a5c90
BTcurve=false
BTfunction=
BTlcd=false
BTprojectFlag=true
BackgroundAlpha=0.2
ChannelTares=0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
Colors=@Variant(\0\0\0\b\0\0\0$\0\0\0\x18\0\x61\0n\0\x61\0l\0y\0s\0i\0s\0m\0\x61\0s\0k\0\0\0\n\0\0\0\xe\0#\0\x62\0\x61\0\x62\0\x61\0\x62\0\x61\0\0\0\xe\0\x61\0u\0\x63\0\x61\0r\0\x65\0\x61\0\0\0\n\0\0\0\xe\0#\0\x37\0\x36\0\x37\0\x36\0\x37\0\x36\0\0\0\x10\0\x61\0u\0\x63\0g\0u\0i\0\x64\0\x65\0\0\0\n\0\0\0\xe\0#\0\x30\0\x63\0\x36\0\x61\0\x61\0\x36\0\0\0\x14\0\x62\0\x61\0\x63\0k\0g\0r\0o\0u\0n\0\x64\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x66\0\x66\0\0\0\x1a\0\x62\0g\0\x65\0v\0\x65\0n\0t\0m\0\x61\0r\0k\0\x65\0r\0\0\0\n\0\0\0\xe\0#\0\x37\0\x66\0\x37\0\x66\0\x37\0\x66\0\0\0\x16\0\x62\0g\0\x65\0v\0\x65\0n\0t\0t\0\x65\0x\0t\0\0\0\n\0\0\0\xe\0#\0\x30\0\x30\0\x30\0\x30\0\x30\0\x30\0\0\0\x4\0\x62\0t\0\0\0\n\0\0\0\xe\0#\0\x30\0\x61\0\x35\0\x63\0\x39\0\x30\0\0\0\f\0\x63\0\x61\0n\0v\0\x61\0s\0\0\0\n\0\0\0\xe\0#\0\x66\0\x38\0\x66\0\x38\0\x66\0\x38\0\0\0\xe\0\x64\0\x65\0l\0t\0\x61\0\x62\0t\0\0\0\n\0\0\0\xe\0#\0\x30\0\x61\0\x35\0\x63\0\x39\0\x30\0\0\0\xe\0\x64\0\x65\0l\0t\0\x61\0\x65\0t\0\0\0\n\0\0\0\xe\0#\0\x63\0\x63\0\x30\0\x66\0\x35\0\x30\0\0\0\x4\0\x65\0t\0\0\0\n\0\0\0\xe\0#\0\x63\0\x63\0\x30\0\x66\0\x35\0\x30\0\0\0\b\0g\0r\0i\0\x64\0\0\0\n\0\0\0\xe\0#\0\x65\0\x35\0\x65\0\x35\0\x65\0\x35\0\0\0\x10\0l\0\x65\0g\0\x65\0n\0\x64\0\x62\0g\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x66\0\x66\0\0\0\x18\0l\0\x65\0g\0\x65\0n\0\x64\0\x62\0o\0r\0\x64\0\x65\0r\0\0\0\n\0\0\0\xe\0#\0\x61\0\x39\0\x61\0\x39\0\x61\0\x39\0\0\0\xe\0m\0\x61\0r\0k\0\x65\0r\0s\0\0\0\n\0\0\0\xe\0#\0\x30\0\x30\0\x30\0\x30\0\x30\0\x30\0\0\0\x10\0m\0\x65\0s\0s\0\x61\0g\0\x65\0s\0\0\0\n\0\0\0\xe\0#\0\x30\0\x30\0\x30\0\x30\0\x30\0\x30\0\0\0\f\0m\0\x65\0t\0\x62\0o\0x\0\0\0\n\0\0\0\xe\0#\0\x63\0\x63\0\x30\0\x66\0\x35\0\x30\0\0\0\xe\0m\0\x65\0t\0t\0\x65\0x\0t\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x66\0\x66\0\0\0\n\0r\0\x65\0\x63\0t\0\x31\0\0\0\n\0\0\0\xe\0#\0\x65\0\x35\0\x65\0\x35\0\x65\0\x35\0\0\0\n\0r\0\x65\0\x63\0t\0\x32\0\0\0\n\0\0\0\xe\0#\0\x62\0\x32\0\x62\0\x32\0\x62\0\x32\0\0\0\n\0r\0\x65\0\x63\0t\0\x33\0\0\0\n\0\0\0\xe\0#\0\x65\0\x35\0\x65\0\x35\0\x65\0\x35\0\0\0\n\0r\0\x65\0\x63\0t\0\x34\0\0\0\n\0\0\0\xe\0#\0\x62\0\x64\0\x65\0\x30\0\x65\0\x65\0\0\0\n\0r\0\x65\0\x63\0t\0\x35\0\0\0\n\0\0\0\xe\0#\0\x64\0\x33\0\x64\0\x33\0\x64\0\x33\0\0\0\x1e\0s\0p\0\x65\0\x63\0i\0\x61\0l\0\x65\0v\0\x65\0n\0t\0\x62\0o\0x\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x35\0\x38\0\x37\0\x31\0\0\0 \0s\0p\0\x65\0\x63\0i\0\x61\0l\0\x65\0v\0\x65\0n\0t\0t\0\x65\0x\0t\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x66\0\x66\0\0\0$\0s\0t\0\x61\0t\0s\0\x61\0n\0\x61\0l\0y\0s\0i\0s\0\x62\0k\0g\0n\0\x64\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x66\0\x66\0\0\0\b\0t\0\x65\0x\0t\0\0\0\n\0\0\0\xe\0#\0\x30\0\x30\0\x30\0\x30\0\x30\0\x30\0\0\0\x12\0t\0i\0m\0\x65\0g\0u\0i\0\x64\0\x65\0\0\0\n\0\0\0\xe\0#\0\x30\0\x61\0\x35\0\x63\0\x39\0\x30\0\0\0\n\0t\0i\0t\0l\0\x65\0\0\0\n\0\0\0\xe\0#\0\x30\0\x63\0\x36\0\x61\0\x61\0\x36\0\0\0\x16\0t\0i\0t\0l\0\x65\0_\0\x66\0o\0\x63\0u\0s\0\0\0\n\0\0\0\xe\0#\0\x63\0\x63\0\x30\0\x66\0\x35\0\x30\0\0\0\x18\0t\0i\0t\0l\0\x65\0_\0h\0i\0\x64\0\x64\0\x65\0n\0\0\0\n\0\0\0\xe\0#\0\x38\0\x30\0\x38\0\x30\0\x38\0\x30\0\0\0\x14\0w\0\x61\0t\0\x65\0r\0m\0\x61\0r\0k\0s\0\0\0\n\0\0\0\xe\0#\0\x66\0\x66\0\x66\0\x66\0\x30\0\x30\0\0\0\f\0x\0l\0\x61\0\x62\0\x65\0l\0\0\0\n\0\0\0\xe\0#\0\x38\0\x30\0\x38\0\x30\0\x38\0\x30\0\0\0\x4\0x\0t\0\0\0\n\0\0\0\xe\0#\0\x34\0\x30\0\x34\0\x30\0\x34\0\x30\0\0\0\f\0y\0l\0\x61\0\x62\0\x65\0l\0\0\0\n\0\0\0\xe\0#\0\x38\0\x30\0\x38\0\x30\0\x38\0\x30\0\0\0\x4\0y\0t\0\0\0\n\0\0\0\xe\0#\0\x34\0\x30\0\x34\0\x30\0\x34\0\x30)
DebugLogLevel=false
Delay=1000
DeltaBTfunction=
DeltaETfunction=
ETBColor=#cc0f50
ETBdeltaColor=#cc0f50
ETcurve=true
ETfunction=
ETlcd=true
ETprojectFlag=true
ExtraEventSamplingDelay=0
Flavors=Acidity, Aftertaste, Clean Cup, Head, Fragrance, Sweetness, Aroma, Balance, Body
KeepON=false
LCDColors=@Variant(\0\0\0\b\0\0\0\b\0\0\0\x4\0\x62\0t\0\0\0\n\0\0\0\xe\0#\0\x30\0\x41\0\x35\0\x43\0\x39\0\x30\0\0\0\xe\0\x64\0\x65\0l\0t\0\x61\0\x62\0t\0\0\0\n\0\0\0\xe\0#\0\x45\0\x42\0\x45\0\x42\0\x45\0\x42\0\0\0\xe\0\x64\0\x65\0l\0t\0\x61\0\x65\0t\0\0\0\n\0\0\0\xe\0#\0\x45\0\x42\0\x45\0\x42\0\x45\0\x42\0\0\0\x4\0\x65\0t\0\0\0\n\0\0\0\xe\0#\0\x63\0\x63\0\x30\0\x66\0\x35\0\x30\0\0\0\xe\0r\0s\0t\0i\0m\0\x65\0r\0\0\0\n\0\0\0\xe\0#\0\x46\0\x38\0\x46\0\x38\0\x46\0\x38\0\0\0 \0s\0l\0o\0w\0\x63\0o\0o\0l\0i\0n\0g\0t\0i\0m\0\x65\0r\0\0\0\n\0\0\0\xe\0#\0\x46\0\x38\0\x46\0\x38\0\x46\0\x38\0\0\0\x4\0s\0v\0\0\0\n\0\0\0\xe\0#\0\x46\0\x38\0\x46\0\x38\0\x46\0\x38\0\0\0\n\0t\0i\0m\0\x65\0r\0\0\0\n\0\0\0\xe\0#\0\x46\0\x38\0\x46\0\x38\0\x46\0\x38)
[... cut ...]
compareBT=true
compareDeltaBT=true
[... 884 lines cut ...]
autoCharge=true
autoDrop=true
chargeTimer=false
chargeTimerPeriod=0
clampEvents=false
eventsGraphflag=2
eventsbuttonflag=0
eventslabelschars=6
eventsshowflag=1
markTP=true
minieventsflags=0, 0, 0
overlappct=100
renderEventsDescr=false
showEtypes=true, true, true, true, true
showeventsonbt=false
specialeventannotations=, , , 
specialeventannovisibilities=0, 0, 0, 0

[grid]
gridalpha=0.2
gridlinestyle=0
gridthickness=1
xgrid=120
ygrid=100
zgrid=10

MAKOMO, 2024-01-08:
Hi Garry,

did you get this running by now?

There are three things todo to get Artisan connect to IKAWA on macOS:

1. run the IKAWA HOME or PRO machine setup from within Artisan (menu Config >> Machine)
2. give the Artisan app permission to access the Bluetooth hardware (apple menu >> System Settings >> Privacy & Security, add entry for Artisan and ensure it is enabled) and turn Bluetooth in macOS on (the IKAWA will not be listed in "My Devices" under Bluetooth as it is a BLE (Bluetooth low energy) device and not a classical Bluetooth accessory like a mouse).
3. turn IKAWA on, place it nearby the laptop running Artisan, click ON in Artisan and it should connect automatically and show readings on the Artisan LCDs

I see on [home-barista](https://www.home-barista.com/roasting/artisan-for-ikawa-home-and-pro-t91360.html) that some are successfully connecting Artisan with their IKAWA. Not sure if they are on Windows or macOS.

drgarys, 2024-01-14:
Thanks, Marko. I was able to connect. Now I’ll have to figure out how to take temperature measurements after I’ve installed removable thermocouples.

MAKOMO, 2024-01-14:
Thanks for this feedback. I will close this issue.

You can simply add an extra device to read your extra TCs (menu Config >> Devices, 2nd tab.

closed 2024-01-14 by MAKOMO