artisan-roaster-scope/artisan #1377: Support for Phidgets VCP4114

Robjones6146, 2024-01-03:
**Please can you add support for the Phidgets VCP4114 clip-on current transducer.**

https://www.phidgets.com/?prodid=1184#Tab_Specifications

You already support VCP1000. VCP1001 and VCP1002 - these measure voltage. I want to measure the amperage of my motors.
The VCP4114 plugs into the Vint HUB and I see no way way of accessing the reading so that it can be used and displayed in the LCD

Maybe there is a way. Have I missed something?

Thank you.

MAKOMO, 2024-01-06:
The sensor issues a 0-5V analog signal which you can read via the corresponding `Phidget HUB IO xx` device with `xx` specifying the HUB port it is connected too (eg. 01 => reads Hub Port 0 and 1 via the two device channels).

Robjones6146, 2024-01-07:
I am using ports 3 and 4. I tried that with no success before. It works now
with 23 but not 45 - will change the ports to match
Also using this formula (x - 2.5) / 0.0625

Thank you.

Robjones6146, 2024-01-07:
Deleted Extra Device as it was not saving the change. Created a new
device, now saving change. Still crashes when I click OFF.

MAKOMO, 2024-01-07:
Which Artisan version are you using? Could you please attach your artisan-settings.aset Artisan settings file (save via menu Help >> Save Settings, rename it to end in .txt and drag-and-drop it here on a comment field on the Github Web page; email attachments will not work!). Thanks!

Robjones6146, 2024-01-07:
Hi Marko,

Sorry, I am confusing things with the other issue. I attached the settings file to the other issue I opened. These issues might all be linked. (https://github.com/artisan-roaster-scope/artisan/issues/1379)
Artisan 2.8.5
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/13853772/artisan-settings.txt)

MAKOMO, 2024-01-07:
Current Artisan version is 2.10.0. Any reason you are using an outdated Artisan?

Robjones6146, 2024-01-07:
No, it's the last version I installed after you made a beta change. I will
upgrade now.

Robjones6146, 2024-01-07:
To be more precise:

You are using a beta continuous build.

You will see a notice here once a new official release is available.

Robjones6146, 2024-01-07:
Updated to 2.10.0

Worse now. Freezing screen. Says Scope monitoring...
OFF and START greyed out

MAKOMO, 2024-01-07:
Your settings are overly complex and your reports are confusing which makes it impossible for me to support you here. Each issue should focus on thing. As I understand, the issue here is on how to read a VCP4114 sensor. I told you that you need to use a `Phidget HUB IO xx` extra device. However, the settings you attached does not contain any extra device of type HUB IO. 

I don't understand your comment
> I am using ports 3 and 4. I tried that with no success before. It works now
with 23 but not 45 - will change the ports to match

at all. To connect a VCP4114 you only need one port, right? So you are either connecting it to port 3 or to port 4. In case you connect the sensor to port 3 you would use an extra device of type `Phidget HUB IO 23` which would also read port 2 (and thus block it for any other use). To connect and block just one single port you can connect that sensor to port 0 of your HUB and use the device of type `Phidget HUB IO 0`. Note that according to your settings you did set the range of the first two IO ports in the Phidgets tab (Config >> Device, tab 4) to +-10mV, but for this type of sensor it should be kept to Auto.

I'd like to suggest to start over and do a factory reset, connect that sensor to port 0 and your TMP1101 to any of the remaining ports, set the main device to TMP1101 4xTC 01 and add one extra device of type `Phidget HUB IO 0`. Turn Artisan on and check the readings you receive on channel one of that first extra device to verify that you get proper readings from your VCP4114.

I am closing this issue again as it is invalid in the sense that Artisan already comes with support for the VCP4114 and your other findings reported here are confusing and there is no clear path on how to reproduce them on our side.

closed 2024-01-07 by MAKOMO

Robjones6146, 2024-01-07:
I forgot to mention that there are 2 x VCP4114
The settings were not saving until I deleted the Extra Device and created a
new one. So the settings I sent you were out of date.