artisan-roaster-scope/artisan #1378: Unable to Toggle Off External PID for Kaleido M6S Pro (New 2023 Version)

cbourne2, 2024-01-03:
**Describe the bug**
I have the new Kaleido M6S Pro (the new 2023 version) that connects to Artisan using Kaleido Serial. I can see within the Artisan code that you reference kaleidoPID (Kaleido's external PID controller), but there isn't a way to modify the PID values for the Kaleido external PID control from the PID control within Artisan. The Artisan PID fields within the Control window do not modify the Kaleido external PID controller. I can also see you have some code commented out to potentially switch between the Kaleido external PID and the built-in Artisan PID controller. It would be great if I could either set the external Kaleido PID values from within Artisan or switch off the Kaleido external PID altogether and use the built-in Artisan PID controller instead. In addition, manually toggling off the Kaleido external PID disables burner functionality between Artisan and Kaleido. The Artisan PID works, but the heating elements in the roaster do not heat up until the external PID function is enabled again within Artisan.

**To Reproduce**
Steps to reproduce the behavior:
1. I manually edited a backup of the settings file to set kaleidoPID=false
2. I then loaded the modified settings to force the external Kaleido PID off so I could use the built-in Artisan PID controller instead
3. The Kaleido roaster heating elements no longer heat up (until the settings are switched back to kaleidoPID=true)

**Expected behavior**
I would like to be able to disable the external Kaleido PID controller and use the Artisan built-in software PID controller instead.

**Setup (please complete the following information):**
 - Artisan Version: v2.10.0 (also tried with the same result on v2.8.4)
 - Artisan Build (number in brackets shown in the about box): c44a180
 - Platform: Windows 11 Pro (OS Build 22621.2861)
 - Connected devices or roasting machine: Kaleido M6S Pro (Nov. 2023)

**Additional context**
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/13816876/artisan-settings.aset.txt)

Terracotta-6, 2024-01-04:
The Kaleido roaster manufacturer's firmware definition does not allow external PID control.

cbourne2, 2024-01-04:
> The Kaleido roaster manufacturer's firmware definition does not allow external PID control.

Thanks for the information! In that case I'll raise this with Kaleido as a feature request.

closed 2024-01-04 by cbourne2