artisan-roaster-scope/artisan #1379: Extra device DAQ1000 detaches for no apparent reason on pressing ON

Robjones6146, 2024-01-03:
**Extra device DAQ1000 detaches for no apparent reason on pressing ON**

The only way to reattach the Humidity sensor (connected to a Phidgets 1136 and then to DAQ1000 and then Vint HUB) is to restart the PC

Artisan 2.8.5 (e969944) 

**Messages:**
The humidity sensor is attached (DAQ1000) -- shows readings in LCD
All the other alarms are activated one by one
Then last of all the Humidity sensor is detached  (DAQ1000) -- LCD reads UU

If I deactivate all the alarms then this does not happen

MAKOMO, 2024-01-06:
Could you please attach your artisan-settings.aset file (save via menu Help >> Save Settings, rename to end in .txt and drag-and-drop it here on the GitHub WebPage to a comment box). Thanks!

Robjones6146, 2024-01-07:
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/13852112/artisan-settings.txt)

Setting file as requested.
Thank you.

MAKOMO, 2024-01-07:
Thanks for your settings file, but don't expect anybody to go over your 50+ alarms, almost 60 button definitions and 4 extra device trying to find out what is going on here without having access to any of your hardware nor an idea what that hardware is all about. Sorry.

Please try to come up with reports that focus on one single issue with a minimal configuration and provide all material like logs, settings and others information (like Artisan version, OS and version, ...) that make it possible to duplicate the issue on our side.

Phidget modules are disconnected by the Phidget driver in low power situations or if they crash due to EMV. Long USB wires are know to cause EMV issues with Phidgets modules, especially if there some strong motors or ignitors running next to them.

Phidgets wrote an excellent article on [Improving Phidgets Hardware Reliability](https://www.phidgets.com/docs/Improving_Phidgets_Hardware_Reliability).

In any case, Artisan is not actively disconnecting Phidget modules on ON or during logging.

You can activate the device logging (menu Config >> Devices, flag "Logging") and check the Phidgets log generated in the log directory

* `~/Library/Application Support/artisan-scope/Artisan` (macOS)
* `C:\Users\<USER>\AppData\Local\artisan-scope\Artisan` (Windows)
* `~/.local/share/artisan-scope/Artisan` (Linux)

I close this issue as it seems to be external to Artisan and rather a hardware issue.

Robjones6146, 2024-01-07:
Thank you.

Thanks for the helpful information. I understand that a lot is going on,
there are 14 Phidgets and lots of automation.