artisan-roaster-scope/artisan #1416: Rename Event Types does not save. Also Save Settings not complete.

2571113, 2024-01-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:
 - Connected devices or roasting machine [e.g. Probatone 5]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2024-01-25:
Your report lags all required information to track your issue(s) down. Especially, it remains unclear on which Artisan version you are experience the issues and on which platform.

The renaming of event types seems to be a duplicate of Issue #572 which has been fixed in the current continuous builds.

What exactly is incomplete for "Save Settings"?

2571113, 2024-01-25:
I have installed the latest version of Artisan.
Artisan 2.10.0 (c44a180)

I rename the event types.
The next time you start Artisan, the old event names will be entered again.

Even saving the settings, resetting Artisan to factory settings and loading a new one does not lead to success.

I can't find a way to rename the event types.

[image]

2571113, 2024-01-25:
I study the #572 and then I install the 2.10.1
And now it works.

Thank you very much!