artisan-roaster-scope/artisan #1418: Latest software version keeps crashing.

gevans3, 2024-01-27:
Artisan crashes out mid roast about every other roast causing me to have to emergency dump my beans, re-start the software, reconnect to the Hot Top, reset the roast and essentially pick up where I left off with what amounts to a heat soak. The problem is, it renders the profile for that particular roast unusable.  I can manually roast to the parameters of my background roast and get close, but doing it twice in a row this morning with $200 worth of Gesha greens is not the way to endear yourself to users.  Artisan 2.8.4 running on a MacBook Air, Sonoma 14.0. Too busy trying to salvage the roast to think about a screen grab but I will do so next time.  Hot Top shuts disconnects and goes into standby and Artisan needs to be re launched and re connected at which point I a launch a new roast that I manage essentially manually based on bean temp and my background telemetry.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:
 - Connected devices or roasting machine [e.g. Probatone 5]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2024-01-27:
Why not use the latest version of Artisan which comes with a fully rewritten Hottop communication layer? Note that your description points to a problem with your physical connection between Artisan and your PC very much like described in Issue #1257, not a software problem. Thus even with an updated Artisan version you might observe the very same behavior. Try to use short USB wires in combination with ferrite coils and move to an environment without much electromagnetic noise (no lamps and other electric switching devices). The malfunction is most likely all about pure hardware design.

gevans3, 2024-01-27:
BINGO. I saw your previous post.  It now occurs to me that I got the
disconnect when I turned on my external bean cooler.  I also have a range
hood running and at one point I turned it up prior to my first crash.  The
first four roasts I had the fan turned down and the bean cooler farther
away.  I repositioned it closer and also turned the range hood fan up a
notch. I have some of the coils on order from Amazon.  Thank you...I will
also check out the latest build.  I had associated my crashes with the
installation of a software update which I did when I moved into our new
house.  It never occurred to me that it wasnt the software but my new
physical setup that was actually causing the problem. THANK YOU.

closed 2024-01-27 by MAKOMO

gevans3, 2024-01-27:
Thanks

George Evans
317.869.3438

*"The completely un-ambitious writer and the really good unpublished poem
are the things we lack most at this time. There is, of course, the problem
of sustenance."*
*
                       -- Evan Shipman*