artisan-roaster-scope/artisan #1432: PID Error Variable as Alarms Source

billysdomain, 2024-02-02:
With a slow response machine like a drum roaster, the PID works best as a PD setup when the error between the BT and SV is very high. When the preheat phase and TP have been processed, I would like to use an Alarm to detect when the PID error is within 5 degrees and then use the Artisan command to apply the integral part back into the PID tune settings, thus returning to full PID mode.

**Describe the solution you'd like**
Add a PID_EV variable that represents the current PID error to the Source drop-down in the Alarms dialog.

**Describe alternatives you've considered**
Pressing the Event button to change the PID tune settings manually

MAKOMO, 2024-02-07:
You can have an extra device channel calculating the delta between the signal your PID is following and current SV (available from the first channel of an extra device of type "PID SV/DUTY %". Consult my post on [Symbolic Formulas](https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html). Then you can have any alarm triggered based on this delta to do what every you think is productive, incl. re-tuning your PID.

billysdomain, 2024-02-08:
Sounds great, are we talking something along these lines:

MAKOMO, 2024-02-08:
Your message might had an attachment (image?) which get's lost if you just reply by email to a GitHub message. Instead please reply on the GitHub Web page to a thread and add your attachments via drag-and-drop to a comment field.

MAKOMO, 2024-02-15:
Closing this as it is not a code issue, but rather a configuration one.

closed 2024-02-15 by MAKOMO