artisan-roaster-scope/artisan #1449: Error installing on MacOS

dapple17, 2024-02-12:
Ok, so as a windows/android user this could totally be on me but I've Googled every way I can think of to solve this issue - MacOS just feels sooo difficult to work around, I just want to download a program and run it

Every time I try to install (I think that's what I'm trying to do?) I get a '"artisan" is damaged and can't be opened. You should eject the disc' prompt. I've tried new downloads, running a pull from the source code through Homebrew on the terminal, reboots...

People keep saying to change the Privacy and Security settings to allow the app to install but I've already allowed app installs from non app store clients

Any help would be appreciated, I've tried figuring out how to download an older version, which is so much easier on a Windows machine, and I can't seem to get anything to work on this one :(

MacOS v. 10.13.6 - 2.3 GHz i5

CarefreeBuzzBuzz, 2024-02-12:
> As a note, I know what a corrupted image is but that shouldn't be happening to ONLY me across dozens of downloads right?

Are you using the legacy build?Aboutartisan-scope.orgMichael Herbert CarefreeBuzzBuzz

dapple17, 2024-02-12:
yes, legacy build; the non legacy throws some sort of incompatible OS prompt as expected

CarefreeBuzzBuzz, 2024-02-12:
Maybe try the continuous build legacy version.  Something may have been fixed that was causing you an issue.  ___________________________Michael Herbert, CarefreeBuzzBuzz

dapple17, 2024-02-12:
Correct, I've tried pulling a few of those and running their txt files through python (in terminal) only to get errors

dapple17, 2024-02-12:
unless someone has a build that you can download floating around out there, I think the routing is incorrect as the files are probably relocated/gone

If this weren't a work computer I'd have partitioned and dual booted into Ubu or something :crying:

MAKOMO, 2024-02-15:
It turned out that the Artisan v2.10.0 macOS legacy build had a signing problem. On start macOS check the code signature and rejects it as faulty. I have resigned the build now correctly. You find the fixed download on the [v2.10 download page](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.10.0) as [artisan-mac-2.10.0-legacy-resigned.dmg](https://github.com/artisan-roaster-scope/artisan/releases/download/v2.10.0/artisan-mac-2.10.0-legacy-resigned.dmg)

Sorry for the troubles.

closed 2024-02-15 by MAKOMO

dapple17, 2024-02-19:
Morning!

Awesome! Thanks for the update, I would gladly run outdated software seeing
as my own hardware is so old but I know how locked in some people can get
while debugging stuff. Anyways I appreciate it!!!