artisan-roaster-scope/artisan #1453: Can an event execute multiple serial commands?

YuenJay, 2024-02-18:
[image]

Only the first command can be executed.

MAKOMO, 2024-02-18:
No. All characters given as documentation to a Serial Command actions are send as one string to the connected device. If the device can interpret this string as separate commands separated by semicolon this is fine. If not you can use a Multiple Event referencing a number of event buttons (using a comma separated list of event button numbers as documentation string) where each of the referenced event buttons issues one of your commands via a corresponding Serial Command action. Note that those event buttons can be hidden and thus with this modeling you have the same effect as if the Serial Command would allow for sequences of commands.

closed 2024-02-18 by MAKOMO

YuenJay, 2024-02-19:
The matter is exactly as MAKOMO said. Thanks very much. If anyone has similar needs as me, the configuration pictures here are for reference.
[image]
[image]

MAKOMO, 2024-02-19:
Thanks for those screenshots. You might want to use the absolute Type "Burner", instead of the relative type "+-Burner", for button 6 and as value the 0 to set the burner slider/event to 0, correspond to your OT1,0 command.