artisan-roaster-scope/artisan #1465: Sliders are gone!

djbb84, 2024-02-24:
## Expected Behavior
Sliders for change gas, drum and damper/fan

## Actual Behavior
sliders is check in dialogue, but they do not appear at anytime. Issue first noticed today 2.24.24, sliders last seen on 2.20.24

## Steps to Reproduce the Problem

  1. Restart roast session/restart Artisan all together
  2. Go into Tools to make sure sliders is check marked.  It shows it is NOT when the roast is off.  But shows it IS selected when stop is ON
  3.

## Specifications

  - Artisan Version: 2.10.0
  - Artisan Build (number in brackets shown in the about box): c44a1802
  - Platform (Mac/Windows/Linux + OS version): MacBook Air 2020, Sonoma 14.2.1
  - Connected devices or roasting machine: Dongyi BY-2 via modbus
[Screenshot.txt](https://github.com/artisan-roaster-scope/artisan/files/14394630/Screenshot.txt)
[Sumatra 32 oz med:dark.csv](https://github.com/artisan-roaster-scope/artisan/files/14394633/Sumatra.32.oz.med.dark.csv)

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

djbb84, 2024-02-24:
> are

djbb84, 2024-02-25:
> Is menu "Views >> Sliders" ticked? What happens if you press the 's' key?

Yes sliders is ticked in menu. When I hit the “s” key, it deflects sliders,
I press s key again and it selects sliders but they still do not appear on
the left side of screen.

djbb84, 2024-02-26:
Sorry hitting “s” deselects sliders

MAKOMO, 2024-02-26:
Could you please attach your Artisan settings. Save them via menu Help >> Save Settings into ´artisan-settings.aset`, rename this file to `artisan-settings.txt` and drag-and-drop it here to a comment box on this GitHub thread (a reply by email with the file attached will not upload it!). Thanks!

djbb84, 2024-02-26:
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/14403449/artisan-settings.txt)

djbb84, 2024-02-26:
Hopefully that worked.

MAKOMO, 2024-02-26:
According to your settings you do not have any of the 4 available sliders activated thus non is displayed.

[image]

closed 2024-02-26 by MAKOMO

djbb84, 2024-02-26:
When I get off work I’ll make sure I see tie slider checked, and then I’ll
recopy the settings file and reshare it.

djbb84, 2024-02-26:
Sorry this file is so big.  Don't know how to take a video and have it be
smaller.
 IMG_6773.MOV
<https://drive.google.com/file/d/16FIrPC2iXuYcjneemGfru-vs-qc9KJAm/view?usp=drive_web>
You should get a link to my google drive

MAKOMO, 2024-02-26:
Open your slider tab (menu Config >> Events, 3rd tab sliders). See my screenshot above. Is any of the flags in the first column ticked to activate a slider? If no slider is active/ticked there is no slider to be shown. Therefore you cannot see any slider and the slider flag in the View menu is greyed (deactivated).

djbb84, 2024-02-26:
They were not ticked. That fixed the issue. Thank you so much!

Is there a shortcut to untuck those boxes? It’s odd that they were
unchecked….

Thanks again! Danke!