artisan-roaster-scope/artisan #1475: IKAWA Home ability to add inlet curve and fan settings to profile?

drgarys, 2024-02-26:
I would like to display the inlet curve as well as BT and exhaust curves for my IKAWA Home, where I’ve installed thermocouples for BT and exhaust.

Artisan does not currently have a way capture the IKAWA Home’s inlet temperature without my installing a thermocouple for that, which would involve disassembling the roaster. But, the IKAWA Home software allows me to program inlet temperature set points, and its PID allows it to follow these set points fairly accurately. Also, the IKAWA Home editing software displays fan settings, which I would like to show in the Artisan recording. 

I attempted to create the inlet curve byI using Designer to input the set points, but I keep losing those set points when I save the profile and try to load it as background. And, I get an unwanted BT curve.

Is there currently a way to program a curve and settings into Artisan and add that info to a profile where I will run the roast or have concluded a roast?

MAKOMO, 2024-02-27:
All HOME IKAWA machines only send out, the PRO two temperature readings over BLE which are recorded by Artisan. Those are named "temp_below" and "temp_above" internally. Note: the second reading was not recorded for PRO machines until v2.10.2 (to be released). All machines send also fan speed and set points (SP) as well as fan and heater level. The PRO machines also send out absolute humidity and ambient pressure readings. All this is recorded by Artisan.

As you know you can record any additional temperature curve by adding extra hardware ,if by Artisan by adding extra device channels.

The IKAWA has a second data structure which is called RoastProfile that contains lists of SPs and Fan settings to control a roast via its internal PID. You can edit such roast profiles using the IKAWA app and send them to the machine for execution. During roasting the machine does not take any control commands via BLE and just follows the selected roast profile stored internally.

We currently have no plans to add an IKAWA roast profile editor to Artisan as the IKAWA app does a good job on this. But as always, Pull Requests are welcome!

drgarys, 2024-02-27:
Hi Marko,

Ah, so the solution is to use Bluetooth to record the inlet temperature and fan in Artisan.

Then you suggest I use extra hardware to input BT and exhaust temperatures. 

Would this work? 

I would plug my BT and exhaust thermocouples into an Amprobe TMD-56 data logger and connect that to my MacBook via USB. 

Gary

MAKOMO, 2024-02-27:
Yes, this should work. You need just to add one extra device of type "Amprobe" to record those extra two channels. If you want to see its BT channel as BT curve you can remap its readings to the main device via a symbolic formula.

I'll close this here as this seems not to be an issue with the implementation but a pure configuration matter.

closed 2024-02-27 by MAKOMO

drgarys, 2024-02-27:
Thank you, Marko.What symbolic formula would I use to map the BT curve to the device? I am not a developer.Gary

MAKOMO, 2024-02-27:
No need to be a developper to map signals in Artisan. The symbolic variable Y1 is bound to ET, Y2 is bound to BT, Y3 is bound to the first channel of the first extra device, Y4 to its second channel, Y5, you guessed it, to the first channel of the second extra device and so on. There is a nice HELP page in the app on this. Just click the HELP button in the extra devices tab or the "Symb ET/BT" tab or read on [Symbolic Formulas: Basics, New Variables and Applications
](https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html)

Assuming your BT signal is recorded by the first extra devices first channel you can just put Y3 in the symbolic formula of the BT (menu Config >> Device, 2nd tab "ET/BT", label "BT Y(x)").

drgarys, 2024-02-27:
Thank you, Marko!