artisan-roaster-scope/artisan #1516: opening completed roast in viewer does not work

dieterc57, 2024-03-18:
### Problem Description

I've had the 'open completed roast in viewer' checkbox ticked in Config->sampling in Artisan v 2.8.4 routinely and it worked fine. Since upgrading to v 2.10.2 this feature does not work for me anymore. Even when opening the viewer manually, the last roast does not load automatically. I have to load it manually each time.
[FZ-04258_24-03-18_1554_Brazil CQ_BR_2021-SB-N-0132_1510.0_091.txt](https://github.com/artisan-roaster-scope/artisan/files/14638462/FZ-04258_24-03-18_1554_Brazil.CQ_BR_2021-SB-N-0132_1510.0_091.txt)
[artisan-settings_20240304.txt](https://github.com/artisan-roaster-scope/artisan/files/14638463/artisan-settings_20240304.txt)

### Steps To Reproduce

Open Config->sampling; check the 'open completed roast in viewer' checkbox and wait for the magic to happen when completing the roast and starting the next one.; it won't :-(

### Artisan Version

2.10.2

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Coffeetech FZ-94 (original, non EVO)

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-03-18:
Thanks for this very detailed report. We could duplicate the issue and fix it. The fix is available as part of the current [continuous builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) and will be released with the next Artisan version.

roasterdave, 2024-03-21:
Some additional changes were made to help this issue.  Will you try the [continuous builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) to be sure this is now working properly?   Thank you for the report.

dieterc57, 2024-03-21:
v2.10.3 works great for me. Thanks for the quick fix.

MAKOMO, 2024-03-22:
Thanks for this confirmation. Artisan v2.10.4, which includes this fix, has now been released.

closed 2024-03-22 by MAKOMO