artisan-roaster-scope/artisan #1521: Contents of Roast Notes is getting deleted

DennieSch, 2024-03-23:
### Problem Description

Contents of 'Roast/Roast Properties/Notes' gets deleted after clicking 'ON'. 
(I first noticed this happening several months back but wasn't sure when it was happening until now.)

### Steps To Reproduce

1. Launch Artisan... v2.10.4 (2078c52)
2. Go to Roast Properties Roast tab and fill in various fields
3. Go to Roast Notes tab and also fill in some pre-roast notes
4. Click 'ON'
5. Go back and look at the Roast Notes tab and the content I previously entered are now gone. 

### Artisan Version

Other - include in Description

### Computer OS and Version

Windows 11 (Surface Pro)

### Connected devices or roasting machine

N/A

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-03-24:
Sorry for this inconvenience. The original behavior (not to delete the roast properties on RESET which is triggered by ON) was changed because a user observed that the note he added for one roast got attached also to all later roasts in the session.

I modified the behavior now to what I think is the intended behavior: if a roast is loaded, doing a RESET (maybe triggered by clicking ON) does clear the roasting notes as it most likely does not apply to the new roast to be recorded. However, on RESET (maybe triggered by clicking ON) the roasting notes are NOT cleared if no profile is loaded at that moment. This allows to enter some notes as preparation before starting the actual roast without having them disappear as you just observed.

Does this make sense?

The (again) changed behavior is already realized by the current continuous build.

DennieSch, 2024-03-24:
Hey Marko,

Yes, that does make sense. I did try it out using CB 3959425 and works as you described. That works great for me and my workflow. Thank you!

closed 2024-03-26 by MAKOMO