artisan-roaster-scope/artisan #1529: Buhler Roastmaster S7 communication error

jaejun78, 2024-04-02:
### Problem Description

While roast progress, it shows error messages below,
======================================================
readActiveRegisters() S7 Communication Error: read_area(5,1061,16,2)@line 341
======================================================

Any other numbers shows well without errors, just error messages shows continuously 
I found this error after 2.10 version.

[image]

### Steps To Reproduce

1. Setup machine buhler roastmaster20 simatic.
2. enter IP address.
3. start.

### Artisan Version

2.10.0

### Computer OS and Version

Windows 11, Windows 10, any other linux machines

### Connected devices or roasting machine

Buhler roastmaster20

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-04-02:
This is a configuration issue. While we implemented the connection to Bühler machines according to the companies specification, it seems that there are some machines (maybe running older firmware versions!?) that do not support the retrieval of the machines state as documented.

The fix is easy. Just deactivate that S7 channel by setting the `Area` of input 6 to the empty entry such that this data is no longer requested from the machine.

[image]

You should also remove those two alarm rules are that refer to this machine state

[image]

We will add a machine setup `RM 20 Simatic Legacy` which will provide those changes with the next Artisan version.

closed 2024-04-02 by MAKOMO

jaejun78, 2024-04-08:
While I asked a dumb question, but great answer!!Thank you so much!!