artisan-roaster-scope/artisan #1539: PID Filters missing

2571113, 2024-04-20:
### Problem Description

Where are the PID filters?
[image]

### Artisan Version

2.10.4

### Computer OS and Version

Windows 11

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

2571113, 2024-04-20:
There were parameters for the PID in the Modbus port settings.

closed 2024-04-20 by 2571113

MAKOMO, 2024-04-21:
Those filters apply only to the internal PID. You have configured Artisan to drive an external MODBUS PID.