artisan-roaster-scope/artisan #1545: Alarm seems not be working correctly

mhsieh66, 2024-04-30:
### Problem Description

The alarm seems not to be working correctly. I set the following alarms but both alarms are triggered at 0:00 after charged, I assume the alarm #9 should be triggered 30 seconds after charged. it looks like using "OR" not "AND" for the time and condition. 
#8	True	0	0	CHARGE	00:00	BT	>	120.0	Slider Burner	0
#9	True	0	0	CHARGE	00:30	BT	>	70.0	Slider Burner	70

### Steps To Reproduce

The alarm seems not to be working correctly. I set the following alarms but both alarms are triggered at the time 0:00 after charged, I assume the alarm #9 should be triggered 30 seconds after charged. the time and condition look like using "OR" not "AND" to trigger the alarm.

#8	True	0	0	CHARGE	00:00	BT	>	120.0	Slider Burner	0
#9	True	0	0	CHARGE	00:30	BT	>	70.0	Slider Burner	70

[Colombia Honey Aponte Hugo Agreda 24-04-30_1058.txt](https://github.com/artisan-roaster-scope/artisan/files/15169438/Colombia.Honey.Aponte.Hugo.Agreda.24-04-30_1058.txt)
[M1S.txt](https://github.com/artisan-roaster-scope/artisan/files/15169443/M1S.txt)
[image]
[image]

### Artisan Version

2.10.4

### Computer OS and Version

macOS 14.4.1

### Connected devices or roasting machine

Kaleido M1

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

Terracotta-6, 2024-04-30:
Process control, not following alert rules.
[image]

mhsieh66, 2024-04-30:
Sorry, you meant my alarm settings don't follow the rules? Could you please explain it? Thanks!

MAKOMO, 2024-05-01:
The time and temperature criteria are indeed combined by a disjunction to decide if an alarm triggers or not. If any of the criteria is fulfilled the alarm triggers. Further, a time criteria of 00:00 is never triggered by  a special convention. Use the almost identical 00:01 instead. To model a disjunction of a time and temperature criteria use an alarm chain.

Please refer to [Alarms](https://artisan-roasterscope.blogspot.com/2013/03/alarms.html) and [More Alarms](https://artisan-roasterscope.blogspot.com/2016/08/more-alarms.html) for further details on the mechanism.

MAKOMO, 2024-05-01:
Try something like this. Alarm 2 triggers 30sec after CHARGE but does not fire any action. Alarm 3 triggers after Alarm 2 has been triggered (by the `If Alarm` condition) and the temperature is higher than 70C and sets the burner to 70%.

[image]

MAKOMO, 2024-05-02:
you intended to say "for most users it is more straight forward that the combination of time and temperature is by conjunction instead of disjunction", right? So you thing + is a more straight forward operation than -? That is just dualism as in the case of disjunction and conjunction. In roasting it is often also the combinator that is used most, stating "if either time or temperature is reached do something".

The artisan alarm rule system as is very achieves the full expressivity of propositional logic (incl. negation!).

closed 2024-05-02 by MAKOMO