artisan-roaster-scope/artisan #1547: not connecting to s7

kentchiong94, 2024-05-07:
### Problem Description

kindly need your help and support because s7 is not connecting, is there any way or to register IP address to connect my artisan to s7?

### Artisan Version

2.10.4

### Computer OS and Version

windows 10

### Connected devices or roasting machine

geisen w15

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-05-07:
Your question is unclear to me. If you run the machine setup for your Giesen (menu Config >> Machine >> WxA) Artisan will ask your for the IP of your roasting machine. Just type it in and connect.

At any time you can change the IP of your machine in the S7 tab (menu Config >> Port) under "Host".

MAKOMO, 2024-05-30:
Any success?

MAKOMO, 2024-06-08:
This issue was closed because it has been stalled for 5 days with no activity.

closed 2024-06-08 by MAKOMO