artisan-roaster-scope/artisan #1548: Kaleido Machine: Temperature controls get unconfigured when a background curve is loaded

ddcprg, 2024-05-10:
### Problem Description

I roast a first batch of coffee from scratch and the save the curve. For the second batch I load the curve I have saved and try to recreate the roast however temperature controls go crazy and can be set properly meaning the roast won't follow the same curve. For example, SV which is the preheat setting for charge temperature is 180C for the first batch and the max is marked a 230C. When the background curve is loaded SV max temp is 120 and it gets loaded to around 90. I'm not sure if this has got anything to do with the fact that I'm using C instead of F but the settings should be kept.

I'm using a Kaleido M10 roaster configured as Kaleido Legacy.

### Artisan Version

2.10.4

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Kaleido M10

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

ddcprg, 2024-05-10:
I'll try to upload some screenshot to this ticket

MAKOMO, 2024-05-11:
Please upload your application settings (menu Help >> Save Settings) as artisan-settings.aset, the profile as .alog file that you loaded into the background as well as the resulting profile recorded.

To upload those files, please rename them to end in `.txt` and drag-and-drop them here on the GitHub Web page into a comment field.

MAKOMO, 2024-06-08:
This issue was closed because it has been stalled for 5 days with no activity.

closed 2024-06-08 by MAKOMO