artisan-roaster-scope/artisan #1565: Artisan v2.10.4 dissconnects and drops roast on Santoker Q7 when using Playback Events on Air settings

Eric18563, 2024-05-25:
### Problem Description

Artisan v2.10.4 dissconnects and drops roast on Santoker Q7 when using Playback Events on Air settings. Usually sometime between 4:15-4:45. I can roast manually just fine, but this has happened every time that I've tried auto events on background roast. 

### Artisan Version

2.10.4

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Santoker Q7

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-05-30:
Could you please attach a background profile which fails as .alog file as well as your settings file as .aset file (menu Help >> Save Settings). Rename both files to end in .txt and attach them here to a comment box on the GitHub page in your browser by a simple drag-and-drop. Note: on simple reply by email attachments are deleted. Thanks!

MAKOMO, 2024-06-08:
This issue was closed because it has been stalled for 5 days with no activity.

closed 2024-06-08 by MAKOMO