artisan-roaster-scope/artisan #1566: macOS communication error

joeky888, 2024-05-25:
### Problem Description

Kapok 501, MODBUS 34 communication error upon connection.

Please help, thanks.
[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/15443487/artisan.log)

### Artisan Version

2.10.4

### Computer OS and Version

macOS 14.5

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-05-30:
Looks like a USB driver or hardware issue. Make sure you download and install the latest CH341 driver from [here](http://www.wch.cn/downloads/CH341SER_MAC_ZIP.html).

joeky888, 2024-05-30:
Solved by forgetting the bluetooth device on macOS and then reconnect again. Thanks anyway.

closed 2024-05-30 by joeky888