artisan-roaster-scope/artisan #1583: Artisan not responding

CaffeineCoffeeRoasters, 2024-06-13:
### Problem Description

Hi!
I renently updated Artisan and set it up for FZ94 from scratch. During the last two times Artisan crashed in the end of the roasts, I think several seconds after I pressed First Crack button. It happened 3 times, with the 2nd and 3rd roasts of the day I guess. The first one or two roasts were ok.
I roast with manual arrangements on the computer (without PID control). I manually drag the sliders of Fan and Drum, switch off the Heating Elements on the roaster.
Settings attached.
Thank you!

[artisan-settings.zip](https://github.com/user-attachments/files/15817048/artisan-settings.zip)

### Artisan Version

2.10.4

### Computer OS and Version

Windows 11 Home 22H2

### Connected devices or roasting machine

Coffee Tech Engineering FZ94

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-06-13:
This is the first crash report since quite some versions. Could you please attach your Artisan log file via Drag-and-Drop to a comment field of this thread here on the GitHub page after renaming it to end with .txt (attachments of replies by email do not work as attachments will be removed by GitHub!).

You find the log file name `artisan.log` on Windows in the following directory

```C:\Users\<USER>\AppData\Local\artisan-scope\Artisan```

Thanks a lot!

CaffeineCoffeeRoasters, 2024-06-13:
I don't see the file you requested.

[image]

CaffeineCoffeeRoasters, 2024-06-13:
do you need this one?

[artisan.txt.log](https://github.com/user-attachments/files/15818225/artisan.txt.log)

MAKOMO, 2024-06-14:
Yep this is the file. Thanks for sharing, but it is not too productive. It shows MODBUS communication issues, but there are no problems on the Artisan side logged, which does not necessarily imply that there are no. Without further indications on what is going wrong it is close to impossible to improve this on the Artisan side. Those communication issues might be relevant.

Your title and your issue description are not well aligned. Your title says "not responding" while your description talks about "crash". What exactly is happening?

Let me give you some suggestions on how to possible improve your situation based on your description and the configuration you attached.

- reduce the MODBUS timeout from 1 seconds to 0.3 or 0.4 seconds (menu Config >> Port, 3rd tab "MODBUS", under section Serial.
- align your Drum quantifier range to the one of the corresponding slider (menu Config >> Events). Set min=28 and max=100. It might be worse to untick those two active quantifiers to deactivate them completely to rule out that those are an issue. As you make your input via the Artisan sliders and buttons and not on the machine this should not hurt. Changes on the machine side will be ignore (however, if I remember correctly those cannot be active in parallel anyhow)
- use a high-quality short USB wire with a ferrit (best on both sides) and, if possible, also add ferrits to the MODBUS/RS485 wiring inside of the machine
- the components of those FZ94 are known to not communicate together well (the frequency drives actual define a slightly different serial settings then those temperature PIDs and wiring them together is a compromise) thus you might not get rid of MODBUS issues completely. However, each version of Artisan contains an improved and more flexible MODBUS communication layer (inherited from the pymodbus project). The current continuous builds of Artisan again use a newer version of this which might improve stability in such border cases. You can download and test that one: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous. A new release of Artisan is expected begin of July which will again be based on a newer version of pymodbus.

CaffeineCoffeeRoasters, 2024-06-17:
Thank you very much for the suggestions. I'll test them on Wednesday.
Also can you please share with me the steps to activate PID control for the Fan and Drum speed controllers? I can't find them in the blog.
Thanks!

MAKOMO, 2024-06-17:
The FZ94 does not feature PID controllers for its fan and drum speed frequency drives. You could apply the Artisan internal software PID to either of those (but only one). However, this does not make much sense to me for either of them as their direct influence on the bean temperature is minimal and thus the expected success minimal. If you want to dry here is a detailed [blog post on the Artisan PID controller](https://artisan-roasterscope.blogspot.com/2016/11/pid-control.html).

What makes more sense to use the [Replay Event](https://artisan-roasterscope.blogspot.com/2017/10/profile-templates.html) feature of Artisan to replay the changes in air flow and/or drum speed as recorded in a previous roast.

CaffeineCoffeeRoasters, 2024-06-20:
Yesterday I was roasting coffee and the program crashed two times again.
Before the roasting I changed the MODBUS timeout from 1 to 0.4 and set the Drum quantifier range 28 to 100 (I now realized that should have aligned it with the Drum slider which is set to 28-70: already changed it now but didn't roast after this change).
I roasted 6 batches yesterday and it crashed on the 6th roast again in the end of the roast. You can see the screenshot attached.
After finishing and switching off the machine I restarted it in several hours and roasted one more batch. The program crashed the same way on that batch again.

[image]

MAKOMO, 2024-06-20:
Did you deactivate the quantifiers to rule those out?

I fear I have not enough information on this case. The log file is empty indicating this is not an Artisan problem. The serial driver might just crash due to EMV effects on your (maybe too long) serial cable. Did you replace the USB cable and/or added ferrite beads to all communication connections? (see the article by Phidgets on some background: https://www.phidgets.com/docs/Improving_Phidgets_Hardware_Reliability)

MAKOMO, 2024-07-05:
Stale

closed 2024-07-05 by MAKOMO