artisan-roaster-scope/artisan #1620: Voice playback in Alarms file not working

vikramms, 2024-07-08:
### Problem Description

I have an "Alarms" file to control my roast and announce some milestones in the profile. I do this using voice.exe and using "Call Program" action. The alarms file is getting stuck at this step and I dont hear any more beeps with the following alarms as the roast moves on. I hear the beep as the steps progress and I also see the text indicating it has reached "Step #" and that is running program "C:/Voice/Voice....." but I dont hear the audio. Also, right there the Alrams file gets stuck and does not move down the profile.

### Artisan Version

2.10.4

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Modbus US Roasters Revelation-23K

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-07-09:
Seems to be an issue with your voice script, external to Artisan. Artisan rules are just calling the script you indicate, everything beyond is up to the OS and the script. Anyhow, you did not attach any information which would allow us to duplicate your issue here.

vikramms, 2024-07-09:
Thanks for the reply. I had to restore 2.4.6 to keep roasting going. I just uninstalled 2.10.5 and re-installed 2.4.6 and my alarm files for various batch sizes worked. So, I don't think it is the arguments in the Call Program lines.

Below is a picture of the Alarm steps.

[image]

I will switch back to 2.10.4 in a day and attach some pictures to show the message that it is stuck on a step.

roasterdave, 2024-07-11:
Thanks for reporting this, it is an interesting one.  There are several text-to-speech applications in use with Artisan, including saystatic.exe, wsay.exe and voice.exe.  I tested all of them and only voice.exe did not work properly.  The voice.exe application quit shortly after being started without outputting any sound.  We haven't seen reports of any other applications that similarly exit in this way.  Reverting a years old change to the environment for external applications resolves this.  

A new [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) will be available in about 30 minutes that works with voice.exe. 

Regarding your report of the alarms getting "stuck," I cannot reproduce this without your settings file. The problem with voice.exe does not cause anything to hang. voice.exe simply quit without making sound.  Please attach your settings *.aset file (Help>> Save settings). If you have a separate alarms file (.alrm) load that first before saving the settings.  Rename the settings file by adding .txt to the end of the file name. Drag-and-drop or copy-paste onto a reply comment field on this website, do not attach to an email.

vikramms, 2024-07-12:
Thanks Dave,
The update definitely fixed voice.exe, but it looks like another issue is broken. I will open another ticket.
But here is my machine setting.
Vikram
[USR25.aset.zip](https://github.com/user-attachments/files/16188255/USR25.aset.zip)

commit 95a61bc2 (2024-07-11): revert subprocess.Popen() redirects to None (Fixes #1620)