artisan-roaster-scope/artisan #1624: Modbus communication issue

vikramms, 2024-07-12:
### Problem Description

The 2.10.5 solved the "Voice/exe" issue but it drew my attention to another problem. First I noticed the temperature readings were off and maybe in "C" even though I selected "F" mode but then the display just got corrupted. Attached are the pictures. The same machine settings file works with 2.4.6, and the Modbus communications is fine. The PLC being used is a Directlogic DL06. However, I am using MODBUS over a UART.
[image]
[image]
[image]
[image]

I am also attaching my settings file.
[USR25.aset.zip](https://github.com/user-attachments/files/16188625/USR25.aset.zip)

BTW, I had the same problem with 10.2.4.

Thank you,

Vikram

### Artisan Version

2.10.5 Continuous Build

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

USRoaster with Modbus

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

vikramms, 2024-07-12:
More screenshots.

Vikram
[image]
[image]

vikramms, 2024-07-12:
2 more

Vikram
[image]
[image]

vikramms, 2024-07-12:
2 more screenshots.

Thanks,
Vikram

[image]
[image]

MAKOMO, 2024-08-15:
Stall

closed 2024-08-15 by MAKOMO

vikramms, 2024-08-15:
Hello Marko,

Sorry, it took me longer than expected. Here are the log files. Also, this situation persists on the new v3.0 release as well.

I was trying to set this up on my Mac and try as opposed to my production PC. However, I am still having several issues with the FTDI driver. Hence the delay.

Thanks,

Vikram

[artisan.log.txt](https://github.com/user-attachments/files/16625986/artisan.log.txt)
[artisanViewer.log.txt](https://github.com/user-attachments/files/16625987/artisanViewer.log.txt)