artisan-roaster-scope/artisan #1627: Willing to help connect my machine with panel from coffeeroasters.com.au

DrR0X-glitch, 2024-07-17:
**Describe the solution you'd like**
I am experienced in python and willing to help get this implemented, just need guidance on how the app works and how I would add to it.

I have a has garanti machine that has been modified with a different mainboard and control panel from coffeeroasters.com.au. I am looking at connecting it to artisan, it currently connects to Coffee Sweet form CRA through a usb cable.

Where would I start with this? Am I missing another way to do this?

MAKOMO, 2024-07-17:
Under menu Devices, there is the option to take the readings from an external program (last option at the bottom). Just point this to your Python script delivering the readings. Press the Help button get the documentation on how your script should return the readings (with Python examples).

If there is more demand for such a configuration you can forward your script to us for integration as regular device.

DrR0X-glitch, 2024-07-18:
Ah perfect, that should work well, I will report back if I'm successful in
reading and controlling the proprietary information.

MAKOMO, 2024-09-20:
Any success?

MAKOMO, 2024-09-23:
Stall

closed 2024-09-23 by MAKOMO

DrR0X-glitch, 2024-10-17:
> Any success?

Sorry for the lack of response, I managed to read some data but couldn't get control of the gas and fan through python due to some encryption and unknown protocols. So just gave up on integration after that. Thanks for your help though!