artisan-roaster-scope/artisan #1658: Fan goes to zero after DROP

RickGroszkiewicz, 2024-08-09:
### Problem Description

I am using artisan 2.10.4 under linux Mint. I have a problem that occurs randomly (oh joy!)

In about 40% of my roasts the fan goes to 0% instead of 100%. I check, and all of the alarm files set the fan to 100% after the DROP occurs. 
[image]
[Costa Rica Tarrazu La Pastora 2022 roasted on 24-07-07.txt](https://github.com/user-attachments/files/16566782/Costa.Rica.Tarrazu.La.Pastora.2022.roasted.on.24-07-07.txt)

### Artisan Version

2.10.4

### Computer OS and Version

linux Mint 21.3

### Connected devices or roasting machine

Hottop B-2K+

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

roasterdave, 2024-08-09:
Looks like everything is working according to the alarm setup.

In your example profile FCs is at 9:26 and DROP at 10:44.  Right after DROP alarm 28 sets the Air to 100.  Then right on schedule at 10:56, which is FCs plus 1:30, alarm 26 fires to set the Air to 0.  

You can try conditioning alarm 26 with "But Not 28" to prevent it firing after alarm 28 has fired.

_Note that the alarm numbers referenced here are taken from the alarm table you posted above.  The alarm table used to record your sample profile was different!  Thus, while the alarm actions remained the same, the alarm numbers in the profile itself do not match with those used here._

[image]

RickGroszkiewicz, 2024-08-09:
Thanks, Dave! I have to remember to prune out extraneous alarms which occur after the DROP ...

closed 2024-08-09 by RickGroszkiewicz