artisan-roaster-scope/artisan #1668: Modbus Communication issue

vikramms, 2024-08-15:
Hello Marko,

Sorry, it took me longer than expected. Here are the log files. Also, this situation persists on the new v3.0 release as well.

I was trying to set this up on my Mac and try as opposed to my production PC. However, I am still having several issues with the FTDI driver. Hence the delay.

Thanks,

Vikram

[artisan.log.txt](https://github.com/user-attachments/files/16625986/artisan.log.txt)
[artisanViewer.log.txt](https://github.com/user-attachments/files/16625987/artisanViewer.log.txt)

_Originally posted by @vikramms in https://github.com/artisan-roaster-scope/artisan/issues/1624#issuecomment-2291331851_

vikramms, 2024-08-15:
Hi MArko,

Another issue can be seen in these 2 pictures.

Even though I set the COM port to 9600 baud, when I exit the window, it still seems to be communicating at 115200.

Thanks,

Vikram

[image]
[image]

MAKOMO, 2024-08-15:
The serial port settings have nothing to do with the MODBUS port settings. According to your attached log file Artisan is trying to connect via MODBUS RTU on COM1 @19200 (not 9600 as in your picture) and does not receive any reply. Maybe the port is wrong or the serial speed, or the register numbers? This seems not to be an Artisan issue. Sorry.

closed 2024-08-15 by MAKOMO

vikramms, 2024-08-15:
Hi Marko,

The pictures were not related to the log file at all. 

The pictures were meant to draw your attention to the fact that, as seen clearly, in the dialog box, I am selecting 9600, but as soon as I close the dialog box, the message on the top left of the graph shows 115200.

But maybe my diagnosis is wrong, it had nothing to do with serial comms or Modbus. Here is the behavior. In the picture below, you can see that both ET and BT values are getting corrupted. If I just uninstall 3.0 and replace it with 2.4.6, the ET/BT LCDs will work just fine. There is no change in settings; everything remains the same.

Regards,

Vikram

[image]
[image]

vikramms, 2024-08-15:
Another log from v3.0 install while looking at the corrupted LCD.

Vikram

[artisan.log.txt](https://github.com/user-attachments/files/16631253/artisan.log.txt)
[artisanViewer.log.txt](https://github.com/user-attachments/files/16631254/artisanViewer.log.txt)