artisan-roaster-scope/artisan #1676: Possible issue with LCD Display

vikramms, 2024-08-19:
### Problem Description

Hi,

I finally got the Modbus working with v3.0 and I can see that the "Scan" button is showing the BT and AT being read correctly as 132.3 and 135.7. 

However, when I click on "ON"/"START" in the logging window. The LCD values show up as "UU". 

Thanks,

Vikram
[image]
[image]
[artisan.log.txt](https://github.com/user-attachments/files/16664506/artisan.log.txt)
[artisanViewer.txt](https://github.com/user-attachments/files/16664507/artisanViewer.txt)

Thanks,

Vikram

### Artisan Version

3.0.0

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

USRoaster with Modbus

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-08-19:
Could you please attach your settings file (save via menu Help >> Save Settings and rename to end in .txt and then drag-and-drop it to a comment field)?

What happens if you untick the "Optimize" flag?

vikramms, 2024-08-20:
Hi,
Here is the .aset file. I will check the optimize flag tomm.
Thanks,
Vikram
[USR25_Art3_0817.aset.txt](https://github.com/user-attachments/files/16668053/USR25_Art3_0817.aset.txt)

vikramms, 2024-08-21:
Hi,
Clearing the "optimize" flag fixed it. It's working now. What does the "optimize" flag do?
Thanks a lot,
Vikram

closed 2024-08-21 by vikramms

MAKOMO, 2024-08-22:
With the MODBUS optimizer active, data is retrieved in blocks from sets of consecutive registers. Your device seems not to support the corresponding (standard!) MODBUS request to retrieve data from more than one register which is needed for this to work.