artisan-roaster-scope/artisan #1685: Cropster XLS import no longer works

smaca69, 2024-08-28:
### Problem Description

Cropster XLS import no longer works because Cropster has renamed the following tabs in their .xls files:

**Curve - Bean temp.**    to    **Curve - Bean temperature**
**Curve - Exhaust temp.**    to    **Curve - Exhaust temperature**
**Curve - Inlet temp.**    to    **Curve - Inlet temperature**

### Steps To Reproduce

Try importing a new Cropster XLS file and it does nothing.  Once you rename the tabs to the old name, it works

### Artisan Version

3.0.2

### Computer OS and Version

Win 11

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

MAKOMO, 2024-08-28:
Could you please attach one of those newer Cropster XLS files (rename it to .txt before drag-and-drop it here on a comment field). Thanks!

MAKOMO, 2024-08-28:
In the best case you could provide an XLS file for each of the languages supported by Cropster....

smaca69, 2024-08-29:
Attachment is a zip file
[Profiles 19_8.txt](https://github.com/user-attachments/files/16807501/Profiles.19_8.txt)

MAKOMO, 2024-09-20:
I updated the Cropster import accordingly. The changes are already in the current [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases).

closed 2024-09-20 by MAKOMO