artisan-roaster-scope/artisan #1694: Modbus conection error

2571113, 2024-09-15:
### Problem Description

With v2.10.4 my Modbus settings worked perfectly.
With v3.0.2 I get Modbus connection errors

What could be the cause?

I have now switched back to v2.10.4 and the MODBUS connection is now working perfectly again.

[image]

### Artisan Version

3.0.2

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

DIY Roaster Arduino Mega

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-09-15:
There seems to be an issue with the updated MODBUS lib underlying Artisan which moved to a more strict interpretation of the MODBUS specification regarding inter character timing which seems to break some of our MODBUS communication over serial.

I tried to work around this. Could you please test and report if the current [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) works any better for you. There is a new "strict" flag in the serial section of the MODBUS tab which is by default not ticked. The new MODBUS lib default is used if it is ticked.

2571113, 2024-09-15:
Thank you for the quick reply.

Unfortunately the Modbus connection doesn't work in 3.0.3 either.
Regardless of whether the "strict-Flad" is True or False

MAKOMO, 2024-09-15:
Hm. Not so good.

In that case I need some detailed logs. Please activate Artisans debug log mode by clicking on the plus icon in the left corner while holding the CTRL and ALT keys. Then also activate the device logging by ticking "Logging" in the Device dialog (menu Config >> Device). Now turn Artisan ON and let it run for some 10sec. Turn Artisan OFF and click again the plus icon, this time holding only the ALT key. An email should open with the log files attached. Send them to the given receiver.

I am quite sure that Artisan v3.0 will work for you as it is still based on the previous MODBUS lib version. Maybe you can confirm this too.

2571113, 2024-09-15:
I have made this logging work around an sent this e-mail.

Now I've been looking for a way to download version 3.0.0.
Tell me, where can you find older versions to download?
I've been looking for a way for 10 minutes...
What's the point of a version history if there's no download available?

2571113, 2024-09-15:
So i can`t find a download of the Artisan V3.0.0 (only sourcecode)

In the V3.0.2 and the V3.03 (with strict-flag) did not work the Modbus connection.
I make this logging workaround an send the protocol by mail

Now I go back to the 2.10.4

Thanks

2571113, 2024-09-16:
These are the Lod files with V2.10.4
There are no problems with the entire roasting process.

[artisan.log](https://github.com/user-attachments/files/17015463/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17015464/artisanViewer.log)

MAKOMO, 2024-09-16:
Those log files don't show the error as there no problems running v2.10.4. To resolve the issue for current later Artisan versions we need a log of the errors which hopefully contain information on how to fix this. For this you need to run the test (with those two debug settings activated) on the problematic Artisan v3.0.2 (or Artisan v3.0.3).

2571113, 2024-09-16:
The Log-files with v3.0.2 are these:

[artisan.log](https://github.com/user-attachments/files/17017238/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17017239/artisanViewer.log)

MAKOMO, 2024-09-16:
Thanks for those logs. Sadly it only states that it never received anything from the device 

```
No response received, Expected 17 bytes Received 0 bytes !!!!
`` 

Are you sure you did configure the serial port here correctly and connected your device while producing this log? Due to the  verbosity of the log settings the "Connect" message is not contained in this file. Again not very useful logs. Sorry.

MAKOMO, 2024-09-16:
The log files also do not seem to complete. They should be exactly 800KB but are only less then half this size. Very strange!

2571113, 2024-09-17:
I have now found the download for v3.0.0.0 and installed this release. 
As you said, the Modbus connection works with this version.
So, I have no idea why it doesn't work with version 3.0.2 or 3.0.3.

Attached are the log files with V3.0.0.0 with some actions on the machine. 

[artisan.log](https://github.com/user-attachments/files/17028531/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17028533/artisanViewer.log)

2571113, 2024-09-18:
Now I have tested the new cotinius build 3.0.3 from roasterdave.
The Modbus connection still doesn't work. Not with the strict flag false or true.
The log files are attached

[artisan.log](https://github.com/user-attachments/files/17049584/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17049586/artisanViewer.log)

MAKOMO, 2024-09-20:
In you logs I only see that Artisan did not receive any thing from your device. 

Could you please download the [latest continuous build v3.0.3](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) again (same name, don't get confused) as I applied some changes that could help to sort this out. On testing, please keep the debug settings ON. THANKS!

2571113, 2024-09-20:
Sorry, it do not work. 
The Connection always switch between 
Verbunden ber MODBUS (5 sec)
MODBUS getrennt (3 sec)

[artisan.log](https://github.com/user-attachments/files/17075590/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17075592/artisanViewer.log)

MAKOMO, 2024-09-20:
Ok. Thanks anyhow for testing. I will take a look at the logs and see if I find something more.
M

2571113, 2024-09-20:
It's not a big problem for me.
I use a very extensive MODBUS communication (~ 30 registers) with Artisan and V.3.0.0 works without problems for my use.

I wonder if another roaster also has a MODBUS problem with V3.0.3
I tried to monitor the MODBUS connection with a serial scanner. The communication with V3.0.0 works perfectly.

With V.3.0.3 there is no communication at all between Artisan and slave (Arduino MEGA 2560).

MAKOMO, 2024-09-21:
There was an issue reported by two users on v3.0.2 regarding serial MODBUS connections caused by an updated MODBUS lib which enforced the strict timing regime from the MODBUS specs. However, I did now work around this one to allow for non-strict timing if that corresponding new flag "Strict" in the serial section is turned off. This resolved the issue for two other users.

The error message
```
09-18 19:09:39 Dummy-2 pymodbus.logging ERROR : could not open port 'COM4': FileNotFoundError(2, 'Das System kann die angegebene Datei nicht finden.', None, 2)
```
indicates that your Windows did not generated the serial com port for your device (or it is not COM4) and thus Artisan could not communicate at all with that serial device. How should it if the specified com port does not exists. This has nothing to do with MODBUS, but with the underlying serial communication and is obviously external to Artisan. 

You should investigate why in your v3.0.0 setup that comport is generated (is it COM4!) and in your v3.0.3 setup not. This must be totally external to Artisan. Could you please verify?

2571113, 2024-09-21:
Port 4 is present, tested and works correctly.

I have now installed Artisan V3.0.3 again and created log files.
I have deleted all previous lines from the log files.

[artisan.log](https://github.com/user-attachments/files/17083543/artisan.log)
[artisanViewer.log](https://github.com/user-attachments/files/17083544/artisanViewer.log)

Both log files only contain data that was recorded after the installation of V3.0.3.

2571113, 2024-09-21:
[artisan.txt](https://github.com/user-attachments/files/17083560/artisan.txt)

MAKOMO, 2024-09-21:
> In the artisanViewer file, I habe found 2 confuse lines:
> 
> 09-18 17:54:09 MainThread artisanlib.main INFO : serial: COM4 @9600
> 09-18 17:54:09 MainThread artisanlib.main INFO : MODBUS Serial RTU: 192.168.5.11, w
> 
> line 1 // right is COM4 @19200
> line 2 // right is COM4 @19200

The first line in your log reads

```
09-21 11:18:02 MainThread artisanlib.util      INFO   : debug logging ON
```

Why not share the full log containing this important information you refer to?

The first of those lines states that the port used for serial communication is set to COM4 @9600. However, this one is totally irrelevant for you as it refers to serial communication (configured in the first tab ET/BT of the ports dialog) and NOT the MODBUS serial communication you are using. 

The second line is about the MODBUS serial communication you are using and which is configured in the MODBUS tab of the ports dialog (3rd tab). This one is configured to comm port COM8. Maybe this is the issue. You should choose on the left side of the MODBUS tab the correct comm port (COM4 in your case!?).

9600, 2024-09-21:
You called?

MAKOMO, 2024-09-21:
> You called?

@9600 : Sorry, forgot to quite this serial baudrate in my comment.

MAKOMO, 2024-09-21:
Wait a moment. You have entries in the ArtisanViewer log file!? Note that Artisan starts in Viewer mode if it detects that there is already one running. The ArtisanViewer is never connecting to any device (hence the name).  Ensure that you have only one Artisan process running, that it is v3.0.3, that the correct serial port is  selected in the MODBUS tab and test with this setup. Everything else is totally confusing.

2571113, 2024-09-21:
So, what is the actual situation?
My hardware (Roaster with arduino mega2560) and the sketch on it works with Artisan V2.0.0 up to V3.0.0 without any problems.
Versions from artisan V3.0.x use a new or different Modbus library.
Result:
The Modbus connection can no longer be established with a version higher than artisan v3.0.0.
We don't have to think about where the error is.

Sorry, but i guess that my software and hardware configuration settings are OK and work with artisan V2.0.0 up to V3.0.0

MAKOMO, 2024-09-21:
The only way we can find out what is going on is trying to duplicate the issue on our side. We have an Arduino mega ordered. Could you please provide us with your Artisan settings (save via menu Help >> Save Settings and attach here renamed to artisan-settings.txt) as well as your Arduino sketch. Thanks!

2571113, 2024-09-21:
Don't be afraid of the Arduino program.
It has been working perfectly for many years.

V3.0.0 is actually enough for roasting but the new versions also have new features in Artisan+
So I can't sleep on V3.0.0 for a few years

[arduino_mega_sketch.txt](https://github.com/user-attachments/files/17085409/arduino_mega_sketch.txt)

[artisan_settings.txt](https://github.com/user-attachments/files/17085410/artisan_settings.txt)

roasterdave, 2024-09-21:
Will you provide a pointer to the correct ModbusRtu library and version used with your sketch?  Thanks.

2571113, 2024-09-22:
The pointer in the sketch to the ModbusRtu.h in here:
 #include <ModbusRtu.h>
 
This is the library
[Modbus-Master-Slave-for-Arduino-master.zip](https://github.com/user-attachments/files/17087838/Modbus-Master-Slave-for-Arduino-master.zip)

roasterdave, 2024-09-22:
Got it, thanks.  Investigating.

2571113, 2024-09-22:
I think we shouldn't waste too much time looking for the problem of my Modbus connection problems.
My DIY roaster works great with my amateur sketch up to artisan V3.0.0. I have implemented everything that is needed in the roasting process. Everything is done by Artisan; heating, bean loading, drum speed, air inlet flap, exhaust flap, exhaust speed, bean temperature, exhaust temperature, bean cooling temperature, PID control, bean ejection, agitator speed, roasted bean ejection flap, etc. I can even roast several batches in a row fully automatically with artisan alarms.

I think the problem is due to my simle unprofessional Arduino sketch. It is therefore difficult to analyze the cause.
I will be releasing a new Arduino code with a somewhat clearer grammar in the next few months. I am confident that this will then be able to communicate with Artisan higher than V3.0.0, or the error can be localized.

On my side we can close this issue. I will let you know if there are any problems with the new sketch.

MAKOMO, 2024-09-23:
We did analyze the issue over the weekend with various hardware and sketches. It turns out that the real issue is the Arduino restarting on opening a serial connection. Depending on the hardware and the sketch this takes a moment such that the first request send from Artisan fails. Sadly on recognizing the failed communication the MODBUS lib underlying Artisan disconnects and reconnects the serial connection which restarts the Arduino again. A large serial timeout could prevent this to happen, but has other negative consequences.

We found the only solution is to introduce a user configurable delay after the serial connect and before sending the first request. For this we retired the current read `Delay` parameter in the serial section of the MODBUS tab and repurposed it to allow the introduction of such a delay directly after a successful connect.

A new [continuous build v3.0.3](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) with this change is now only. We would love if you could give this a test using the following settings in MODBUS tab and give us feedback.

- Timeout: 0.4s
- Delay: about 1.6s (as low as possible, but as high as needed)
- Retries: 0

Two notes on your sketch and your MODBUS configuration:
1) there are several `delay()` calls in your code, within `setup()` and `loop()`. Removing those if not needed for anything might speed up the readiness for communication after a restart (which always happen on serial connect)
2) the data which is requested by Artisan per sampling is not organized in a continuous block as the registers 7 to 10 are not read. Thus Artisan either sends two requests if `fetch full blocks` is not ticked (one request for registers 1-6 reading 12 bytes and another request for register 11 and 12 reading another 4 bytes) or if `fetch full blocks` is ticked just one request avoiding the additional overhead of a second request, but reading register 1-12 (24bytes) while only the data of 8 registers (16bytes) is used. This can be improved by relocating the data of registers 11 and 12 to the registers 7 and 8, and resulting in a faster communication, maybe even allowing for a 2sec sampling rate.

2571113, 2024-09-23:
Unfortunately, I can't establish a Modbus connection with the current build V3.0.3 either.

I have tried various combinations of

- Timeout: 0.4s
- Delay: about 1.6s (as low as possible, but as high as needed)
- Retries: 0

Unfortunately, a Modbus connection was not possible with any combination.

With V3.0.0 I even have a connection with this setting:
Timeout: 1s
Delay: about 0s
Retries: 0

But if I go with Timeout: 0.9, then no connection is possible, no matter what I enter for Delay or Retries.

roasterdave, 2024-09-23:
This is strange.  I am able to solidly connect from Windows 11 to an Arduino Mega running your sketch (with the sensor code stripped since I don't have that hardware and the temperatures set constant, original delays intact).

Using your settings file with:
- Timeout: 0.4s
- Delay: 1.0
- Retries: 0

Would you please attach your current artisan.log file (no need for artisanViewer.log) so we can take a look?

2571113, 2024-09-23:
This is the log file from this evening, with various attempts.
[artisan.log](https://github.com/user-attachments/files/17101734/artisan.log)

After testing I go back to V3.0.0

I will now installing V.0.3 on my Windows Surface and trying your settings:

Timeout: 0.4s
Delay: 1.0
Retries: 0

After that I'll send feedback with the log file.

It takes 30 minutes

2571113, 2024-09-23:
So, now I have install the continuous build V3.0.3 onto the Windows surface. This uses COM 3 for Modbus communication.

I have set these settings:
Timeout: 0.4s
Delay: 1.0
Retries: 0

Unfortunately, NO connection to the machine (arduino mega) is possible.
The log file is attached

[artisan.log](https://github.com/user-attachments/files/17102497/artisan.log)

roasterdave, 2024-09-23:
The log shows Artisan trying to open MODBUS on COM4, not COM3.  Please double check the setting on the Modbus tab.  Thanks.

2571113, 2024-09-23:
The COM3 is right. Arduine IDE con connect the board on COM3

Another current log file

[artisan.log](https://github.com/user-attachments/files/17102944/artisan.log)

and the Modbus settings
[image]

MAKOMO, 2024-09-23:
Did you try with a higher Delay, say 3sec. As I said, as high as needed, but as low as possible. It might take your  Arduino longer to restart due to the other code you have around, like for the display and such.

roasterdave, 2024-09-23:
Okay, communication is taking place now on COM3.  It looks like this startup delay is not long enough for your setup.  I'd increase the delay to 3 and set retries to 2.  If that still fails try a longer delay.

MAKOMO, 2024-09-23:
Note that the 'Delay' parameters on 3.0.0 and 3.0.3 are different and serve different purposes. Only on the current v3.0.3 that parameter adds a delay (in seconds) after the connect to wait until the Arduino fully restarted and can start answering requests. On all versions before, that parameter only added a (small) extra delay (in milliseconds) after the last request.

2571113, 2024-09-23:
In V3.0.0 I can set various parameter combinations. The Modbus connection always works.

In V3.0.3 it is the other way round. No matter what values ​​I use, no Modbus connection is established.

I think the problem is with the pymodbus library introduced in V3.0.2. Are there any other "Modbus" roasters that have already switched to V3.0.2? Is the Modbus connection working for anyone?

MAKOMO, 2024-10-02:
The issue you encountered with Artisan v3.0.2 was caused by the update of the pymodbus lib to v3.7.x not supporting retries on empty answers any longer ([remove unneeded client parameters](https://github.com/pymodbus-dev/pymodbus/pull/2272): remove `retry_on_empty`, see [the pymodbus Changelog](https://github.com/pymodbus-dev/pymodbus/blob/dev/CHANGELOG.rst)). Not a problem for a regular MODBUS client, but for clients not answering the first request after connect like your Arduino (due to its restart on connect) the Artisan level retry mechanism never succeeds as  pymodbus just closes the connection on errors and thus Artisan did reopen it on next retry with the same negative result.

We added a delay after connect before sending the first request for serial connections which can resolve this issue.

You might want to check out the [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous). It is based on a more modern async variant of the same MODBUS lib which allows for retries without disconnects. This allows to connect to devices like your Arduino which drop requests the first seconds after a connect due to internal delays (a restart in your case).

This update will be part of the next Artisan version.

closed 2024-10-02 by MAKOMO

2571113, 2024-10-02:
Ok, thank you very much for the information. I will test the Coniius bulds and try different delay times. I will also try to optimize the Arduino code for the new situation.

2571113, 2024-10-02:
Update: I loaded the new continuous build from today. And with this setting below the Modbus connection now works without interruption. Thanks for the support!

[image]