artisan-roaster-scope/artisan #1695: Alarms did not trigger (based on BT)

RickGroszkiewicz, 2024-09-15:
### Problem Description

Something strange happened in one of my recent roasts. Several alarms should trigger after the BT reaches 270 F. But three alarms did not trigger, as shown in the screen shot.

### Steps To Reproduce

Loaded roasting profile, and started roast
[image]
[Colombia El Porvenir Pink Bourbon 2023 roasted on 24-09-08.alog.txt](https://github.com/user-attachments/files/17006954/Colombia.El.Porvenir.Pink.Bourbon.2023.roasted.on.24-09-08.alog.txt)

### Artisan Version

2.10.4

### Computer OS and Version

linux Mint 21.3

### Connected devices or roasting machine

Hottop B-2K+

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-09-15:
Hm. The circled BT is @253C, but the alarms condition is BT > 320 (thus that BT is still lower than the set limit, right). Looks all good to me or am I missing something?

RickGroszkiewicz, 2024-09-15:
I thought alarms 7, 9 and 10 should have triggered before alarm 11.

MAKOMO, 2024-09-16:
Alarms are not triggered in order. The order is irrelevant, only their conditions count. Alarms 7,9 and 10 did not trigger as their time condition is disabled (00:00) and the temperature condition `BT>x` not fulfilled as the BT never exceeded x yet. While the temperature condition is also not yet fulfilled for Alarm 11, its time condition `1min after TP` was already fulfilled at this point. Note that the time and temperature conditions of each rule are connected by a disjunction. Thus whichever condition of those two holds first triggers the rule action.

RickGroszkiewicz, 2024-09-16:
The time condition was supposed to be 0:00 for alarm 11. Looks like a random keystroke made this happen!

closed 2024-09-16 by RickGroszkiewicz