artisan-roaster-scope/artisan #1697: Mac (Intel) Monterey 12.5 No generic serial devices shown

dprotter, 2024-09-17:
### Problem Description

I am trying to send serial commands to an arduino (no specific communication protocol). I am able to connect to the Arduino and send messages through the arduino IDE. However, the artisan Port tab for generic serial com ports is completely unpopulated. As you can see, somewhere Artisan is configured to send over COM4, and is trying to send serial data. 

Interestingly, other tabs DO recognize the arduino (such as the MODBUS tab). Any help would be appreciated!

Finally, I AM capable of communicating over serial if I save the config file, manually assign the correct port in a text editor, and load the settings. Not the worst solution! but a bit inconvenient. 

[image]
[image]
[image]
[image]
[image]

### Steps To Reproduce

1. start artisan
2. attach arduino
3. open "ports" tab

### Artisan Version

3.0.2

### Computer OS and Version

Mac OS 12.5.1 (Intel)

### Connected devices or roasting machine

arduino

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-09-17:
The "Extra" tab of the Ports dialog is to configure serial ports for extra devices. You might not have configured any extra device in the Device Assignment dialog (menu Config >> Device, 2nd tab).

The serial port of the main device is configured in the first tab ET/BT of the Ports dialog.

MAKOMO, 2024-09-20:
Could you resolve your issue?

MAKOMO, 2024-09-23:
Stall

closed 2024-09-23 by MAKOMO