artisan-roaster-scope/artisan #1700: An unhandled exception occurred. Please report the problem on Github: https://github.com/artisan-roaster-scope/artisan/issues  When reporting this issue, please include your settings file (export  via menu Help >> Save Settings) and the details below.  An entry has been written to the error log (menu Help >> Error).  Version: 3.0.3, revision: 50dc3b3 2024-09-25, 19:54:15 -------------------------------------------------------------------------------- : invalid literal for int() with base 10: '' (line: 22826)

2571113, 2024-09-25:
### Problem Description

Ther is an error when I change some PID settings

### Artisan Version

3.0.3 Continuous Build

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

2571113, 2024-09-26:
[image]
If you change the PID settings registers and try to save them, you will receive an error message and the changes will not be saved.

MAKOMO, 2024-09-26:
Thanks for your report. We could duplicate the issue by leaving fields empty where numbers are expected. Commit 6a5ccefb78b186bc9ae0f060089454fac5f6efb3 should make the data input more robust. 

Does the current [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) fixes this for you? Could you please test?

Thanks!

2571113, 2024-09-26:
That's right. If you make the changes one by one using the Tab key, the changes and saving will work.

MAKOMO, 2024-09-26:
So is this issue now fixed in the new Continuous Build or not?

2571113, 2024-09-26:
The new on is 45 minutes old.
I will try it tomorrow

2571113, 2024-09-26:
the current build is ok.

MAKOMO, 2024-09-26:
Thanks for testing and this confirmation!

closed 2024-09-26 by MAKOMO