artisan-roaster-scope/artisan #1702: I have a Santoker R3 Im trying to connect to artisan

airwavecoffee, 2024-09-29:
I recently bought a used santoker r3 
I downloaded the santoker app on iPad, it works but not a huge fan.

I want to connect to artisan via usb. I have downloaded different usb drivers and used my Mac and a pc
cannot figure out how to connect. what am I doing wrong?

MAKOMO, 2024-10-17:
[Santoker R Series support](https://artisan-scope.org/machines/santoker/) was just improved. Could you please check if the setup `R Series Bluetooth` (menu `Config >> Machines >> Santoker`) of the current [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) works with your machine (assuming it is not a R Master Series and it features Bluetooth communication).

MAKOMO, 2024-11-06:
Stall

closed 2024-11-06 by MAKOMO

MonkeyBeast0113, 2025-04-22:
Hello, 

I had the same issue connecting my R3. Did you figure it out? I’m using windows. Thanks

MAKOMO, 2025-04-23:
The next Artisan version will improve Santoker support and should work fine with the R3. Either test the current Continuous Build or wait until the end of the month for the next version.