artisan-roaster-scope/artisan #1704: Cant save any Palette changes

Robjones6146, 2024-10-01:
### Problem Description

I cant save any Palette changes. I am using the latest Artisan version 3.0.2 I cant think what else to try, this does not make any sense other than some kind of conflict.

I have three Palettes which I saved in a previous version (not sure when or what version.)
I restarted PC, Restarted Artisan a few times.

[alog_file.txt](https://github.com/user-attachments/files/17200971/alog_file.txt)
[settings_file.txt](https://github.com/user-attachments/files/17200972/settings_file.txt)

Same for previous version (2.10).

### Artisan Version

3.0.2

### Computer OS and Version

Windows 10

### Connected devices or roasting machine

Many Phidgets

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [ ] No files are attached.

Robjones6146, 2024-10-01:
> Trying to understand and reproduce your issue. I loaded your setting file
> (but not the alog profile) and there are no palettes defined. Then I load
> your palette file via the Palette tab, close the Events dialog and see 3
> palettes in use: #0 (many buttons visible), #1
> <https://github.com/artisan-roaster-scope/artisan/issues/1> (some buttons
> visible), and #2
> <https://github.com/artisan-roaster-scope/artisan/issues/2> (one red
> button visible). I can switch between them using the CTRL+number keys.
>
> Now I choose palette #2
> <https://github.com/artisan-roaster-scope/artisan/issues/2>, open the
> Events dialog and added a TEST button as button #59
> <https://github.com/artisan-roaster-scope/artisan/issues/59> (57 buttons
> hidden!). Then I switch to the Palette tab and store this modified palette
> under name TEST as palette #3
> <https://github.com/artisan-roaster-scope/artisan/issues/3>. I leave the
> events dialog and can switch between those palettes incl. the new palette
> #3 <https://github.com/artisan-roaster-scope/artisan/issues/3>.
>
> I go back to the Events dialog, Palette tab and click save to save the new
> palette state including my new palette.
>
> Now I do a factory reset, load again your settings and load this palette
> file. I close the events dialog again and are back to my 3 palettes which I
> can toggle using the number keys.
>
> Nothing wrong that I could spot here. Did you press "Store Palette" after
> making changes?
>
> Could you please given me detailed step-by-step instructions demonstrating
> your issue starting from a factory reset and loading your artisan settings?

Thanks for your reply.

If you take a look at the all_text.txt file you can see that the there is
only  ', 'palette(1)', 'palette(2)'],

I just figured out that I can't overwrite a .apal file but I can create a
new one and the test button now works. See the newPalette file.

, 'palette(1)', 'palette(2)', 'palette(0)'],

'palette(1)', 'palette(2)', 'palette(3)'],

{'0': [[4, 4, 4, 4, 4, 4, 4, 4, 1, 0, 2, 2, 2, 2, 4, 4, 4, 4, 4, 4, 4, 4, 4, 1, 1, 0, 4, 4, 9, 9, 9, 9, 9, 4, 4, 4, 4, 4, 2, 9, 9, 9, 9, 9, 4, 4, 4, 4, 9, 4, 4, 4, 4, 4, 4, 4, 4, 4, 4], [0.0, 0.0, 1.1, 1.1, 1.1, 0.0, 1.1, 1.0, 2.0, 2.0, 3.0, 7.0, 8.0, 9.0, 0.0, -1.1, 1.1, -1.1, 1.1, 0.0, 1.1, -1.1, 1.1, 11.0, 0.0, 0.0, 1.1, 0.0, 0.0, 0.0, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1, 1.1], [20, 3, 3, 3, 6, 6, 6, 6, 6, 6, 14, 14, 14, 14, 3, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 0, 0, 0, 2, 2, 2, 2, 20, 14, 0, 0, 2, 2, 2, 6, 6, 6, 6, 2, 3, 2, 2, 2, 2, 2, 2, 20, 20, 20], [1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1], ['clearBackground ; loadBackground(//REDSERVER/Artisan files/genericA1.alog)     # Loads the last generic file created by FM across all 3 roasters', '3, sleep(30) ,4      #   , 38    ---   Load ON - delay 30 sec -Load OFF -- button(START) or 38   -- Click Load then also starts the roast after 30 second load cycle', '16, 17, 18  #Roast chamber slide open, Transport blower ON, Transport Hopper open', '19, 20, 21   #Roast chamber slide close, Transport blower OFF, Transport Hopper close', 'set(1,1,684949:3) # Burner on  --  REL1000', 'set(1,0,684949:3) # Burner off  --  REL1000', 'set(0,1,684949:3) #Blower VFD on --  REL1000', 'set(0,0,684949:3) #Blower FVD off  --  REL1000', 'vel(1, 0.15 ,684949:1) # 3v at pos 10 - Belimo cold Damper slightly open', 'vel(0, 0.15,684949:1) #0.15 = 3v - Value according to slider - Belimo hot Damper slightly open', 'out(0,2,684949:4) # VFD signal 2 volts - Blower lowest speed   **NEVER LOWER than 30%**', 'out(0,6,684949:4) #Blower  OUT1002   -- Blower VFD speed control voltage 6v', 'out(0,7,684949:4) #Blower  OUT1002   -- Blower VFD speed control voltage 7v', 'out(0,8,684949:4) #Blower  OUT1002   -- Blower VFD speed control voltage 8v', '6, 8, 25, 26, 39, 46, 48, 56   #Multi event -- 6 Burner off, 8 Blower off, 25 Cold full closed, 26 Hot full closed, 39 Zero Loft, 46 chaff off, 48 airlock off, 56 batch', 'vel(0,-1,684949:2) #Bean in roast chamber  slide open  (Slide normally closed)', 'set(0,1,687340:2) #Transport Blower motor on  --  REL1000  -- activated using alarm when Load beans - 4', 'vel(0,-1,687340:1)  #Green beans hopper slide open --  DCC1000 -- activated using alarm when Load beans - 4', 'vel(0,1,684949:2)  #Bean in roast chamber slide close 1 minute later', 'set(0,0,687340:2) #Transport Blower  motor off --  REL1000  -- activated using alarm 1 min after Load beans - 7', 'vel(0,1,687340:1)  #Hopper slide close --  DCC1000 -- activated using alarm 1 min after Load beans - 7', 'vel(1,-1,684949:2) # Linear actuator  open --   DCC1003 ', 'vel(1,1,684949:2) # Linear actuator   close --  DCC1003', 'vel(1, 0.5 ,684949:1) # 0.5 = 10v (full open) Value according to slider  - Belimo cold air Damper full open', 'vel(1, 0.1 ,684949:1) #0.1 = 2v (full closed) -  Belimo cold air Damper full closed', 'vel(0, 0.1 ,684949:1) # 0.1 = 2v (full closed) Value according to slider - Belimo hot air Damper full closed', 'set(3,1,684949:3) # Water open  --  REL1000 ', 'set(3,0,684949:3) # Water closed  --  REL1000  -- use sleep(1) one second per 5kg', '# Send beans to Tub', '# Send beans to Weighfil', 'Caramelization begins at @ ~180C', 'Degrades @ ~192C', 'Water quench at DROP', 'C:/Program Files/wsay.exe "Burner on" --voice 2 --speed 45 #Burner on', 'C:/Program Files/wsay.exe "Blower on" --voice 2 --speed 45 #Blower on', 'C:/Program Files/wsay.exe "Aproaching first crack!" --voice 2 --speed 45', 'C:/Program Files/wsay/wsay.exe "Roast complete!" --voice 2 --speed 48', 'button(START)', 'out(0,0,684949:4) #Blower  OUT1002   -- Blower VFD speed control voltage 0v', 'Bean Outlet Closed 30 seconda after opening at COOL END', 'Bean Outlet Opened at COOL END', 'C:/Program Files/wsay.exe "Beans too hot .... reduce hot air" --voice 4 --speed 48    #@~240C ', 'C:/Program Files/wsay.exe "Beans burnt .... reduce hot air" --voice 4 --speed 48     #@~250C', 'C:/Program Files/wsay.exe "Air too hot .... reduce hot air" --voice 4 --speed 48    #@~260C', 'set(2,1,684949:3) # Chaff motor ON  --  REL1000 (shared relay)', 'set(2,0,684949:3) # Chaff motor OFF  --  REL1000 (shared relay)', 'set(2,1,684949:3) # Rotary airlock motor ON  --  REL1000  (shared relay)', 'set(2,0,684949:3) # Rotary airlock motor OFF  --  REL1000  (shared relay)', 'C:/Program Files/wsay/wsay.exe "Inlet above 270 degrees .... reduce temperature ... reduce temperature" --voice 4 --speed 48    #@~270C ', '27, sleep(1), 28    # Water on -- sleep 1 second -- water off', 'C:/Users/Redber/Documents/profileA1.bat      #The background roast profile has been loaded', 'C:/Users/Redber/Documents/loadA1.bat      #The beans are loading', 'C:/Users/Redber/Documents/startA1.bat      #Beans have loaded >> Roast has started', 'C:/Users/Redber/Documents/endA1.bat      #The Roast is now complete', 'C:/Users/Redber/Documents/onA1.bat      #The Burner,  [... cut ...]  100, 100, 100], [0, 0, 0, 0], [0, 0, 0, 0], ['', '', '', ''], [0, 0, 0, 0], '', [0, 0, 0, 0], [0, 0, 0, 0]], '9': [[], [], [], [], [], [], [], [], [], [0, 0, 0, 0], [0, 0, 0, 0], ['', '', '', ''], [0.0, 0.0, 0.0, 0.0], [1.0, 1.0, 1.0, 1.0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [100, 100, 100, 100], [0, 0, 0, 0], [0, 0, 0, 0], [100, 100, 100, 100], [0, 0, 0, 0], [0, 0, 0, 0], ['', '', '', ''], [0, 0, 0, 0], '', [0, 0, 0, 0], [0, 0, 0, 0]], 'maxlen': [20, 30, 30, 30, 14, 14, 14, 14, 14, 14]}

Robjones6146, 2024-10-01:
[newPalette.txt](https://github.com/user-attachments/files/17202122/newPalette.txt)

Thanks for your reply.

If you take a look at the all_text.txt file you can see that the there is only  ', 'palette(1)', 'palette(2)'],
I can create a button and it appears, but it does not work and it does not save.

I just figured out that I can't overwrite a .apal file but I can create a new one and the test button now works. See the newPalette.txt file.
Now it works and is saved.
Can you reproduce this issue?

, 'palette(1)', 'palette(2)', 'palette(0)'],
'palette(1)', 'palette(2)', 'palette(3)'],

Thank you.
Robin

Robjones6146, 2024-10-01:
I have now got it to work.

I had to clear   *#1 default *  by clicking *<<Store Palette*
 with current field *blanc*.

Then I repeated the process but this time with the *#1  *clicked  *<<Store
Palette  *with the current field  *default* ... now it saved correctly.

MAKOMO, 2024-10-01:
I am now totally confused. Is there now an issue? What exactly is the issue and what are the steps from Factory Reset to reproduce them. If there is no issue, please close this issue. Thanks!

Robjones6146, 2024-10-01:
The issue is described in the previous post.
Artisan (Artisan on my PC) does not overwrite a previous stored palette unless you clear the name and then Store it with the same name or a new name. 

The whole saving of Palettes is very confusing (to me).

MAKOMO, 2024-10-01:
> The issue is described in the previous post. Artisan (Artisan on my PC) does not overwrite a previous stored palette unless you clear the name and then Store it with the same name or a new name.

Yes, but I cannot duplicate it. That's why I asked for the exact steps that would allow me to reproduce this. Once I can reproduce the issue I might be able to fix it. Otherwise it will get hard;)

Note that the label under "current:" is just the name of the last activated palette (if any) and it is used to "tag" a palette on "Store Palette". Those tags are listed in the palette popup after their number. By activating a palette, changing its label, and storing it in the same position you can change the label of that palette. However, those labels are just tags and do not prevent or enable the storing of the current events configuration in a palette slot.

> The whole saving of Palettes is very confusing (to me).
I am sorry. But, you are invited to improve the system at any time. Just submit a PR with your changes to the code and we will review it. Remember this is a volunteer project depending on YOU!

Robjones6146, 2024-10-01:
Thanks Marko,

No complaints from me. I really appreciate the software and work done. My coding skills are not that advanced. I tried so many things and combinations of things, before and after I upgraded to the latest version and then did the same things again. When I cleared the tag attached to the #0 or #1 and then renamed it, only then did it work and then it saved when I saved the Palette. 

 If this sound confusing, it is.

closed 2024-10-01 by Robjones6146