artisan-roaster-scope/artisan #1705: Kaleido Bluetooth Dropping Between Roasts (Unable to Reconnect without Roaster+Laptop Reset)

jessevanderhoff, 2024-10-03:
### Problem Description

[artisan-settings.aset.txt](https://github.com/user-attachments/files/17239143/artisan-settings.aset.txt)
[24-10-02_2113.alog.txt](https://github.com/user-attachments/files/17239144/24-10-02_2113.alog.txt)
[image]
[https://youtu.be/Zrp8efzq0Fg](url)

Kaleido M6 Sealed connects to Artisan via bluetooth successfully but then disconnects from M1 MacBook Air after ending roast and cannot reconnect thereafter without full roaster and Artisan laptop reset. 

Youtube video link of issue is posted

### Steps To Reproduce

1. Restart roaster and Artisan MacBook laptop
2. Start Artisan 
3. Connect to roaster via Macbook bluetooth menu
4. Machine is already configured in Artisan from previous session.
5. Turn roaster on through Artisan
6. Run simulated roast 
7. End roast through Artisan "Off". "Kaleido disconnected" is displayed
8. Click "reset" in Artisan
9. Confirm Comm Port is maintained via "Config", "Port"
10. Click "On" in Artisan
11. See uu in ET,BT.

Youtube video here. [https://youtu.be/Zrp8efzq0Fg](url)

### Artisan Version

3.0.2

### Computer OS and Version

MacOS Sequoia 15.0

### Connected devices or roasting machine

Kaleido M6 Sealed

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [X] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

Terracotta-6, 2024-10-03:
Cannot share the same aset, M6, M1 Bluetooth pair with the coffee roaster and save separately.

MAKOMO, 2024-10-03:
Thanks for your detailed description of this issue. Thus we cannot duplicate it here. There seems to be a problem on the OS level with the bluetooth connection. I see in your video that the bluetooth device never gets into the "connected" state in the Bluetooth setup window of macOS and it keeps turning (despite the connection initially seems to work). Not much we can do on the Artisan level as Artisan is just connecting to a serial port and if that port is not available for connect this will fail. Could it be a macOS 15.0 Bluetooth issue?

Anybody else with this issue?

Terracotta-6, 2024-10-03:
Control circuit adopts MCU EPS32 Bluetooth, compatibility is best to consult the equipment supplier after sale.
https://www.kaleidosniper.com

WesFarley, 2024-10-06:
@MAKOMO I am also having this issue.  My connection has started to decay during the roast.  See the attached profile showing the gaps in the roast timeline as well as my configuration settings.  I cannot reestablish a connection to the roaster without endless pairing unpairing and restarting/toggling on artisan until I am eventually successful which is fairly nerve racking when I'm trying not to warp my drum.  I am running windows 11 so a different OS than @jessevanderhoff.

[Attachments-Connection Issues on my M2.zip](https://github.com/user-attachments/files/17270984/Attachments-Connection.Issues.on.my.M2.zip)

I sent an email to espresso outlet (US distributor) who thought it looked like a software vs hardware issue on the roasters end, for whatever that perspective may be worth.

MAKOMO, 2024-10-07:
@WesFarley: your profile indicates that your BT connection is unstable as it keeps dropping. This might have several reasons like disturbing signals from motors nearby, large distance between sender and receiver, broken antenna,... However, Artisan manages to reconnect to the machine again until the next drop. Nothing that can be improved on the Artisan side here. Artisan just connects to a serial port and if this breaks tries to reconnect. This seems to work as intended in your case. Thus your issue seems unrelated to the one reported by @jessevanderhoff.

WesFarley, 2024-10-07:
@MAKOMO, of the factors you mentioned, I cannot find a fitting cause.  The machine is only 10ft from the controlling computer, and I do not have dropout issues with any other BT devices.  It's of course possible that the issue is different than @jessevanderhoff but the behavior seems to be very similar.

I'll be going hard wired barring another fix.

jessevanderhoff, 2024-10-08:
@MAKOMO @Terracotta-6 is it possible to download an old version of Artisan to test this issue? 

I'm particularly looking for v2.8.4 as it is the first version with Kaleido support built in. 

The US distributor for Kaleido roasters, Espresso Outlet indicated they believe this could be an Artisan issue specific to new versions. 

Being far from software savvy I found the [release page](https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.4) and source code but don't see a DMG file. 

[image]

Terracotta-6, 2024-10-09:
Kaleido   Enterprise microdisk  craftsman- Mac-2.8.4 (Mac computer).dmg

https://shfront.wxwork.qq.com/downloadobject?fileid=08011204313330332210313638383835343539353336343039302a01313223343462496c41585751654f52596f636d5636454f367730305f31363931383236333838388fb0f38e014214945a61432e9d19d7275237ffafa4c2ea4ff7cfcd48015802600768b8177207333030303030309001d4f1dca6069a0104636f7079a0019ed203&filename=artisan-mac-2.8.4(Mac%E7%94%B5%E8%84%91).dmg&authkey=700800100118702270e6ff394bba5810de36226dfc4a796e28f2c51d447e53575007df7a4909cb6f66d7809d9330d3dfbf8807a4732fe6e0c74e7718994e8ea34db20cb63c99a8859dd81c1f87c02628fdeb418617e6c3742a22d56e941141d8dd321337929dc7a990f622965a78a112e0d0960d23a9a97b6b&weixinnum=13102670058671489

Terracotta-6, 2024-10-09:
@jessevanderhoff   I am a Kaleido Coffee Roaster user, don't know what you bought?  PRO?  Dual system？you can download Mac-2.8.4 and try.
[User guide.pdf](https://github.com/user-attachments/files/17303158/User.guide.pdf)

huangdaojiri, 2024-10-09:
Contact your dealer to replace the roaster’s main board

cnladd, 2024-10-18:
> Thanks for your detailed description of this issue. Thus we cannot duplicate it here. There seems to be a problem on the OS level with the bluetooth connection. I see in your video that the [...]

I am having the same issue just as described by the original poster. Kaleido M10 Pro roaster, MacOS 15.0 Sequoia. Turn on roaster, MacBook Air connects to roaster via Bluetooth, turn on roaster with Artisan, roast. Turn off roaster. Reset roaster. No connection on next roast. The only sure fix so far is to reboot my laptop in between each and every roast.

I haven't had a chance to try on an older version of MacOS yet. I will this weekend.

jessevanderhoff, 2024-10-19:
> I am having the same issue just as described by the original poster. Kaleido M10 Pro roaster, MacOS 15.0 Sequoia. Turn on roaster, MacBook Air connects to roaster via Bluetooth, turn on roaster with [...]

I have tried a few different combinations and found the following:

1. Older Mac OS versions with Apple Silicone exhibit the same issue.
2. Older Mac OS versions with Intel Silicon work as intended. 
3. For those with "dual" machines and a Kaleido tablet, a wifi connection can be used and maintains connection through multiple roasts. The only caveat is that the roaster will not store the wifi settings and the tablet must be connected during roaster boot up to initiate the connection. 

I've contacted Kaleido who have been provided minimal help. I recommend to email them and their US distributor, espresso outlet to put pressure on them to fix this issue. 
kaleidosniper@yahoo.com
sales@espressooutlet.com

Terracotta-6, 2024-10-20:
Blueman is a GTK+ Bluetooth Manager
https://github.com/blueman-project/blueman/tree/2.4.3

Subteran, 2024-12-03:
I have the same issue. It started with macOS Sonoma and has continued with Sequoia. MacBook Pro M1 Max. After doing a roast, hit off, hit reset, save profile. Quit Artisan, open Bluetooth Settings, forget Roaster, turn off Bluetooth, turn on Bluetooth, connect to Roaster, close Bluetooth Settings, restart Artisan, turn On, connects and displays Temp data. If I do not do this, no temp data will display when I select On.

MAKOMO, 2024-12-04:
@Subteran : not totally unrelated. Bleak is used for BLE communication. The Kaleido is using a regular serial connection to a port provided by the OS (potential via a classic Bluetooth connection). Sorry.

morsecoder, 2024-12-20:
Was just looking at the bluetooth trace from my Macbook Air M1 / macOS Sequoia 15.1. It seems like the initial handshake for RFCOMM Connection Request works just fine and the RFCOMM Connection Disconnect after roast seems successful as well. When clicking Reset/On to start another roast, I don't see any new connection requests being sent at all. Wondering if this helps at all. Also, is there a way to run the Artisan application in debug mode so I can see if there are any sus application logs? 

This could still just be an issue between OS/Kaleido, but curious if there's something we can do to reset the connection that doesn't involve "Forgetting device" and reconnection.

morsecoder, 2024-12-20:
Actually seems like it's timing out trying to reconnect to the serial port.

12-20 13:29:00 Thread-2 (start_background_loop) artisanlib.kaleido   DEBUG  : connecting to /dev/cu.Roaster_SRSv6@57600 ...
12-20 13:29:01 Thread-2 (start_background_loop) artisanlib.kaleido   DEBUG  : ping response timeout
12-20 13:29:03 Thread-2 (start_background_loop) artisanlib.kaleido   DEBUG  : ping response timeout
12-20 13:29:05 Thread-2 (start_background_loop) artisanlib.kaleido   DEBUG  : ping response timeout
12-20 13:29:06 Thread-2 (start_background_loop) artisanlib.kaleido   DEBUG  : connection timeout

MAKOMO, 2024-12-20:
John,

I made a minor modification to the async code which better ensures that the transport gets closed on disconnect. Not sure this makes a difference. The change is in current [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous). Maybe you find time to test this.

To toggle debug mode in Artisan on, press OPTION and COMMAND while clicking the plus icon in the upper left corner of the main window. Hold OPTION while clicking that icon opens an email with the current artisan.log file attached. There is a logging flag in the Devices dialog (menu Config >> Devices) to turn device logging on. However, I have no high hope that there are trace of the issue found in that log file as the troubles might be outside of Artisan.

Thanks for looking into this!

morsecoder, 2024-12-20:
Just tried it out and it looks like it now freezes when selecting "OFF". Seems like await writer.wait_closed() might be getting stuck based on whatever writer.close() is doing (or not doing).

MAKOMO, 2024-12-21:
So somehow that closing is not successful and that is why it cannot reopen that port as it still not closed from previous connection. That wait_closed() is just awaiting the proper closing of the port before continuing now. So something must be broken with that Bluetooth port closing on recent macOS running on AppleSilicon. We never got an issue report for Linux, Windows or Intel Macs. Not sure how to find out what is really broken here. Artisan is using very high-level APIs and the issue seems to be down that stack somewhere. Any help is very appreciated!

MAKOMO, 2024-12-21:
I added a timeout around my change and now this should at least not freeze anymore. I checked with a simulated (non-bluetooth) serial port (for the lack of a Kaildo machine) and the port is correctly closed (after the closing down message is correctly send to the machine) and reopened. Not much more I can do on the Artisan side. Seems that the underlying Bluetooth connection initiated first is not correctly closed on closing the serial port on Apple Silicon machines and a new one is not created on opening that serial port again.

morsecoder, 2024-12-21:
Thanks for looking into this! Agree that it looks like an underlying Bluetooth issues, I'm seeing the Disconnect Response from the machine but OS probably isn't parsing it correctly or something. I might look into it a bit more but found a quicker workaround is to just kill the bluetooth process with `sudo pkill bluetoothd` and is restarts and reconnects haha.

danielcroft, 2025-01-06:
For another point of reference; I'm seeing this issue in the most recent stable release of Artisan (3.1.0)
```Artisan 3.1.0 (adaee2a96) 
Python 3.12.7, Qt 6.7.3, PyQt 6.7.1, Matplotlib 3.9.2, NumPy 2.1.3, SciPy 1.14.1, pymodbus 3.7.4 (1.20.20240911), Yoctopuce 2.0.62875 (2.0.62875) 
https://artisan-scope.org/ 
Core Developers
Rafael Cobo, Marko Luther & Dave Baxter 
License
[GNU Public Licence (GPLv3.0)](http://www.gnu.org/copyleft/gpl.html)
```

`MacOS Sequoia 15.1.1 / Apple Silicon`

The only thing I've got working is to remove the bluetooth connection for the Kaleido M2 and re-add it (no other changes) at which point it starts working again.

Reproduction steps:
1. Connect to Kaleido M2 in bluetooth settings
2. Click "on" (BT & ET should read valid numbers)
3. Click "off" (wait, it takes a second to time out)
4. Clicking "on" again yields no valid temperatures

Work around:
1. After pressing the "off" button
2. Go to bluetooth settings
3. Remove your Kaleido roaster
5. Re-add your Kaleido roaster
6. Click "on"

EdGoesBoom, 2025-01-19:
Is this reproducible with MacOS Sequoia 15.2 / Apple Silicon  (Apple M1). ?

jessevanderhoff, 2025-01-27:
> Is this reproducible with MacOS Sequoia 15.2 / Apple Silicon (Apple M1). ?

I was able to reproduce on my laptop. 

Artisan Version

3.1.0

Computer OS and Version

MacOS Sequoia 15.2

Processor

M1 Pro

Connected devices or roasting machine

Kaleido M6 Sealed

I connected with bluetooth successful upon initial power up
-I then ran a dummy "roast" with fan and drum only. 
-Ended the roast in Artisan and saved the roast and settings file. 
-Reset via Artisan and attempted to connect to Kaleido again but failed. I saved a second dummy roast and settings file in case thats helpful. 

Files ending with "_2" are the second, failed connection "roast".

[artisan-settings.aset.txt](https://github.com/user-attachments/files/18552613/artisan-settings.aset.txt)
[artisan-settings_2.aset.txt](https://github.com/user-attachments/files/18552611/artisan-settings_2.aset.txt)
[25-01-26_1744_2.alog.txt](https://github.com/user-attachments/files/18552612/25-01-26_1744_2.alog.txt)
[25-01-26_1741.alog.txt](https://github.com/user-attachments/files/18552610/25-01-26_1741.alog.txt)

EdGoesBoom, 2025-02-04:
I ran a test with my Apple Silicon Mac to see if I could get anything interesting in the artisan.log file with Help->Serial->Log = ON if I run 'sudo pkill bluetoothd' in between the roasts.

Artisan 3.1.0; Apple M1 Pro Sequoia 15.3; New Kaleido M2 Dual 

There are two tests in the log file.  The test procedure is:

Click ON button and check if bluetooth connects (successful in both tests)
Click OFF button
(Baseline test do nothing / Second test run "sudo pkill bluetoothd")
Click ON button and check bluetooth connection (First test is not successful / Second test with sudo is successful)
Click OFF button

The artisan.log file is annotated with "TEST:" at the beginning of line where I've "echo 'TEST: $msg' >> artisan.log".

[artisan.log](https://github.com/user-attachments/files/18650398/artisan.log)

Some observations:
1. When the roaster does not connect, artisan is sending pings with no response.
2. When the roaster does connect, after 1st ping there are series of artisan writes followed by handshaking between the roaster and Artisan.
3. In the second test when artisan reconnects after killing bluetooth the /dev/cu.Roaster_SRSv6 file is not found.

Not sure if this is helpful.  Let me know if there are additional tests that would help or if there is a question.

MAKOMO, 2025-02-04:
@EdGoesBoom : thanks for your log file.

There seems to be a problem to release the serial port on OFF. It is unclear if this is a problem of Artisan, macOS or the roasting machine. As that port is therefore not available on second connect, the connect fails. Artisan still tries to send a ping (or at least logs that it tries to send), however, that fails as there is no active connection. If on ON a working connection to a serial port can be established, Artisan does some handshaking with the machine to take over the control.

As I do not have access to any Kaleido machine I cannot run any experiments to find out what goes wrong here. As all goes well on most platforms, like macOS on Intel and Windows, I can only guess that something is wrong on macOS on Apple Silicon. Not necessary a bug in the OS, but some "incompatibility" in the large software stack between the Artisan UI and the bluetooth chip. 

Most likely something goes wrong on disconnect which cause the serial port (or bluetooth connection) not to be released correctly. I found one thing experimenting with my software Kaleido simulator. Artisan seems to release the connection before the result of the very last command "release control" that it sends to the machine got full received (for some wired technical reasons). I did correct this one, however, I have no high hope that this fixes the issue.

Maybe you find time to test the current [Continuous Release](https://github.com/artisan-roaster-scope/artisan/releases) and see if this works any different. Note that those builds are not signed nor notarized by Apple. Thus your first start needs to be via a right-click on the app icon selecting "Open".

EdGoesBoom, 2025-02-04:
The continuous release had same behavior.  I've attached the log file as before.

Are you ok with the write: {[CL AR]} after stop sampling?   I was expecting to see [CL AR SN].

[artisan_311_20260204.log](https://github.com/user-attachments/files/18660295/artisan_311_20260204.log)

More observations that I have are:

I monitored the System Settings for Bluetooth.  On the first roast I see the bluetooth connect on Mode ON and disconnect on Mode OFF as expected.  When making the second roast Mode ON the bluetooth for the roaster is not connecting also as you would expect as it doesn't connect. the important bit is that the bluetooth connection is off according to the system.

Since I leave the Kaleido on during my test I'm going to assume that the source of the issue is not the Kaleido hardware.

Once I've done a roast, if I quit the Artisan application and then restart the application the bluetooth connection to the roaster does not connect (unless I do a sudo). If I do a kill -9 on the Artisan process and then restart the bluetooth connection to the roaster does not connect.

Subteran, 2025-02-04:
Same behavior for me also.

MAKOMO, 2025-02-05:
> The continuous release had same behavior. I've attached the log file as before.
>
> Are you ok with the write: {[CL AR]} after stop sampling? I was expecting to see [CL AR SN]. [...]

Looking at your log file, I don't understand exactly what you did. To me it seems that you initially connected to the roaster then clicked OFF and Artisan was sending the `write: {[CL AR]}` (no SN in the request according to the protocol, only in the reply!) which was correctly acknowledged by a `received: {0,SN:00000001}`. 

Then you turned Artisan ON again and it Artisan is not receiving any data from the machine any more, but it looks like it did connect to that serial port successfully. Just that this serial port is no longer properly connected to the bluetooth connection towards the machine.

I could imagine that either the machine is not releasing the bluetooth connection correctly (which is unlikely as it works if accessed from Artisan on other platforms like macOS on Intel) or that macOS is not correctly releasing the bluetooth connection the moment Artisan is releasing the serial port. I am quite sure that Artisan correctly release the serial port as it otherwise couldn't connect again to my serial port emulation. As Artisan has no idea of a bluetooth connection underneath the serial port, thus I don't see how this can be fixed on the Artisan side.

Any ideas?

EdGoesBoom, 2025-02-06:
I've got a few ideas about the cause and an idea about a better flow.

I'd bet a cheese sandwich that the issue lies somewhere in the Python asyncio or modbus packages or some unexpected/unintended side affect from how you are using it. I did a search online and there was some chatter about garbage collection in the async package but that seemed to have been resolved some time ago.  Not sure if there are other Python applications that are doing similar to what you are doing. Not sure what Python, asyncio, modbus versions are used to build Artisan and if they are up to latest.  It would make sense to try make a simple standalone test that could be passed to those groups.

Regarding a better flow.... When doing back to back roasts we seem to only have OFF button option to end recording but then all of the control buttons go away.  But I still want to be able to control the roaster to turn off the cooling fan or whatever. Is there a more streamlined way to stop recording, but then update the roast (log roasted weight and make notes), save the roast log etc... but still keep the Bluetooth connection open for controlling the roaster and then doing my back to back roasting protocol before starting logging the next roast?   Should I make that a separate request?

MAKOMO, 2025-02-06:
> I've got a few ideas about the cause and an idea about a better flow.
> 
> I'd bet a cheese sandwich that the issue lies somewhere in the Python asyncio or modbus packages or some unexpected/unintended side affect from how you are using it. 

MODBUS is not involved here

> I did a search online and there was some chatter about garbage collection in the async package but that seemed to have been resolved some time ago. Not sure if there are other Python applications that are doing similar to what you are doing. 

The asyncio package is used virtually for all communication in Python by now.

> Not sure what Python, asyncio, modbus versions are used to build Artisan and if they are up to latest. 

Check the `requirements.txt`. Artisan is always build with the latest stable releases of all libs.

> It would make sense to try make a simple standalone test that could be passed to those groups.

Go! Please report on your findings here.

> Regarding a better flow.... When doing back to back roasts we seem to only have OFF button option to end recording but then all of the control buttons go away. But I still want to be able to control [...]

This is unrelated and was discussed several times (use the search box of GitHub above) and will not change. You can tick "Keep ON" under menu `Config >> Sampling` to have Artisan automatically turn ON again after OFF.

MAKOMO, 2025-02-06:
> > It would make sense to try make a simple standalone test that could be passed to those groups.
> 
> Go! Please report on your findings here.

Please note that all works absolute fine on all platforms (incl. macOS on Intel) but for macOS on Apple Silicon.

EdGoesBoom, 2025-02-06:
Disappointing that the allure of a cheese sandwich is not enough but I'd answer same in your shoes.  ;)

I don't understand how using PyModbus.transport.sertransport doesn't involve modbus somehow but that's for me and my crusty Python skills to try and determine. If I can make progress I will report here.

Will take the flow for back to back roasting to /dev/null as I understand your comment.

Appreciate all you do for the community.

MAKOMO, 2025-02-06:
> I don't understand how using PyModbus.transport.sertransport doesn't involve modbus somehow but that's for me and my crusty Python skills to try and determine. If I can make progress I will report here.

Artisan uses a fix applied by pymodbus to serial-async which is no longer maintained. Instead making a copy of it, Artisan just links to it as it is anyhow importing pymodbus.

Instead of investigating on the Artisan side, I would rather investigate, why re-opening the serial port does not re-open the bluetooth port on macOS / AppleSilicon. MIght be

EdGoesBoom, 2025-02-06:
The approach I was thinking of was to write a standalone Python (maybe copying a fair amount of kaleido.py) but with a main() that exercises the Bluetooth in a way that might expose the issue. I may not get to it for a while.  Let me know if you disagree.

quasistoic, 2025-02-07:
> This is unrelated and was discussed several times (use the search box of GitHub above) and will not change. You can tick "Keep ON" under menu `Config >> Sampling` to have Artisan automatically turn ON again after OFF.
> 

I used the search box but did not succeed in finding any of the relevant discussions. If you believe this would be an unnecessary tangent in this case, could you please link the relevant issue(s)?

Thanks.

roasterdave, 2025-02-07:
>This is unrelated and was discussed several times (use the search box of GitHub above) and will not change. You can tick "Keep ON" under menu Config >> Sampling to have Artisan automatically turn ON again after OFF.

>I used the search box but did not succeed in finding any of the relevant discussions. If you believe this would be an unnecessary tangent in this case, could you please link the relevant issue(s)?

Have a look at this [blog post](https://artisan-roasterscope.blogspot.com/2020/06/working-together-artisan-artisanviewer.html) that includes information about "Keep On."

Conversation about "Keep On" and related workflow are not relevant to this Bluetooth issue.  For further questions or discussion please open a [New Discussion](https://github.com/artisan-roaster-scope/artisan/discussions/new/choose) item.

peterghaddad, 2025-02-07:
Also running into the same problem and on a Mac M1 with a Kaleido M10. Have to disconnect bluetooth after each roast... Are there plans to prioritize this? Can I help provide logs or relevant information?

MAKOMO, 2025-02-08:
The Artisan team has no access to any Kaleido machine to work on this and see also nothing on the Artisan side that goes wrong here. Thus the Artisan team is currently not active on this.

quasistoic, 2025-02-08:
Is anyone on the team in the San Francisco Bay Area? I'd be happy to help them get hands on a machine here if it would help.

roasterdave, 2025-02-10:
@quasistoic  That is a gracious offer, thank you.  I do live in the Bay Area though I'm out of town this week. Happy to connect.  Rather than post contact details here, send them via the [contact form](https://artisan-roasterscope.blogspot.com/p/contact-me.html).  But before getting too excited be aware that I'm a Windows user without access to a Mac with Apple silicon.  That is where we suspect the problem resides, as described above.

quasistoic, 2025-02-10:
> @quasistoic  That is a gracious offer, thank you.  I do live in the Bay Area though I'm out of town this week. Happy to connect.  Rather than post contact details here, send them via the [contact [...]

Contact info submitted. I have a Kaleido M6 and Apple M1 hardware that exhibits the problem behavior. Happy to work with you on it when you're available.

EdGoesBoom, 2025-02-11:
I'm also Bay Area (Marin) with Apple Silicon and M2 roaster.  I've been tinkering with a standalone Python that tries to recreate the issue but my (limited) Python skills are slowing me down. I also submitted contact info.

In the meantime I figured another work around using Alarms.

Alarm triggers when "ON".... After 0:01.  Action is "Call Program".... program is "/usr/bin/sudo /usr/bin/pkill bluetoothd"

The tricky part is to edit the /etc/sudoers file so that a password is not required to call pkill bluetoothd. You should know how to use the 'vi' program for editing a file.

% sudo visudo /etc/sudoers

Add a line that looks like the following but replace "EdGoesBoom" with your MacOS username:

EdGoesBoom ALL = (ALL) NOPASSWD: /usr/bin/pkill bluetoothd

BL9PE, 2025-02-16:
Hey EdGoesBoom - thanks for this innovative solution!  I've been struggling with this exact issue on a MBP M3 since upgrading to Sequoia 15.3.1 - that's over Bluetooth to a Kaledo M2 Sealed.  The call to restart the BT process takes about 10s and then its golden!

danielcroft, 2025-02-16:
This worked for me as well. I also tried setting this on the reset event which works but this happens on both pressing the reset button as well as when you click the on button.

CarefreeBuzzBuzz, 2025-02-16:
With this Alarm solution, have you considered running it on OFF rather than ON.  As I understand the problem, could that be cleaner so that it connects immediately with ON.  This assumes no Between Batch Protocol is running.

BL9PE, 2025-02-17:
Yes - off would be cleaner, but I dont see an option in the alarm triggers to set this?

With a bit more testing, I'm finding that I do seem to need to restart Artisan for the roaster to reconnect.  Artisan hangs for 20s or so before it quits if its previously been connected successfully to the roaster

MAKOMO, 2025-02-17:
@BL9PE : Each button can have an associated command incl. the OFF button (no need for an alarm rule). See menu Config >> Events, first tab.

dsawali, 2025-02-26:
Hey there @BL9PE, I'm having this exact same issue as yours. I have a Kaleido M6 Dual with an M4 MBP and am struggling to get this to work properly. Do you mind elaborating on the solution a bit more? 

How do I utilize events to restart the BT connection? Cheers

EdGoesBoom, 2025-02-26:
@dsawali Review backwards on this thread in Github to [this message](https://github.com/artisan-roaster-scope/artisan/issues/1705#issuecomment-2649634897). I prefer to restart BT when pushing ON button because I never know what was going on previously.  

Summary is to create an alarm in Artisan that kills the bluetooth daemon. The tricky bit is to use the visudo command in a terminal on your Mac.

Subteran, 2025-02-26:
@dswali since you have a dual you can use a WiFi connection

https://youtu.be/1t97xRpt_bY?si=lTU6i7jlDysbApi_

dsawali, 2025-03-10:
@Subteran thank you for reminding me. I do have the dual, and it does have a wifi capabilities. But when I tried to search for wifi networks, it won't capture my home wifi. It detects all of my neighbours' wifi though. I'm not sure what could be the issue here. Perhaps it's a 2.4/5Ghz situation?

Terracotta-6, 2025-03-11:
WIFI 2.4GHz connection must meet the following conditions, 1. MCU EPS32 model ,  2. Firmware versions are indispensable.
https://products.espressif.com/#/product-selector?language=zh&names=

danielcroft, 2025-03-11:
And if your network name is a bit weird, it’s not interested.

blacksheepaul, 2025-08-01:
> I'm also Bay Area (Marin) with Apple Silicon and M2 roaster. I've been tinkering with a standalone Python that tries to recreate the issue but my (limited) Python skills are slowing me down. I also [...]

sudo /usr/bin/pkill bluetoothd, then it works, twice ...

john-unger, 2025-10-02:
I'm encountering this issue as well. with the aid of ChatGPT it was researching and said this:
> this isn’t a Kaleido-only problem—on macOS (esp. 15 “Sequoia”) the OS leaves the SPP (Bluetooth serial) session in a “busy”/half-torn-down state for a bit after you close it. In Artisan 3.0.x, the Kaleido driver tears the link down when you press Off, and the immediate reopen on On hits that stale state: you get uu for ET/BT and cannot recover without power-cycling the roaster and/or macOS’ BT stack. The proper fix is to (a) stop closing the serial link on Off for Kaleido Bluetooth, and (b) make the serial open/close on macOS more robust: prefer /dev/cu.* over /dev/tty.*, avoid exclusive locking, and ensure a clean shutdown of the reader thread before closing.

beanoccio, 2025-10-02:
ChatGPD can probably also tell you the source of this information.

edwin-truthsearch-io, 2026-08-27:
For anyone landing here with a Kaleido (or another Bluetooth SPP machine) on macOS: I ran into a neighbouring symptom and opened #2226 with a log analysis.

During a roast my Mac entered idle standby — after first crack, waiting for second crack, no keyboard or trackpad input for several minutes — the Bluetooth SPP link was torn down and the machine started its cooling cycle on its own. Artisan does not inhibit the macOS idle system sleep while it is ON; the only power related call is `appnope.nope()` at startup, which uses `NSActivityUserInitiatedAllowingIdleSystemSleep` and thus explicitly *permits* the idle system sleep.

What my `artisan.log` shows afterwards is what this thread describes: once the link is gone the device node itself disappears and the connect loop retries `open()` on a path that no longer exists.

```
08-23 21:39:04 artisanlib.kaleido ERROR : read failed: [Errno 6] Device not configured
08-23 21:39:05 artisanlib.kaleido ERROR : [Errno 2] could not open port /dev/cu.Roaster_SRSv6: No such file or directory
08-23 21:39:10 artisanlib.kaleido ERROR : [Errno 2] could not open port /dev/cu.Roaster_SRSv6: No such file or directory
```

PR #2225 inhibits the idle system sleep while Artisan is ON (macOS, Windows and Linux), reconnects the machine on wake for suspensions that cannot be inhibited such as closing the lid, and logs the connection lifecycle at the default log level so a drop is visible in `artisan.log` without enabling debug logging first.

To be clear about the scope: it does **not** fix the vanished device node discussed in this thread — that still needs a fix at the Bluetooth/macOS level — but it should prevent one common way of getting into that state in the first place.

edwin-truthsearch-io, 2026-09-06:
Two findings from measuring this on macOS 26.6.2 (Apple Silicon, Kaleido over Bluetooth SPP) that may save people in this thread some time.

**Toggling Bluetooth is not what fixes it.** With the port in the stale state I compared the device node before and after a full Bluetooth off/on cycle: it survived with the *same* minor number, and it stayed silent. What actually rebuilds the serial device is **removing the pairing and pairing again** — unpairing deletes `/dev/cu.<machine>` outright, and the next connect succeeds in about two seconds. The Bluetooth toggle only makes the unpaired machine visible again so it can be paired, which is likely why it became part of everyone's ritual.

**`pkill bluetoothd` no longer works here.** On this version the daemon is protected: both `pkill` and `launchctl kickstart -k system/com.apple.bluetoothd` are refused with `Operation not permitted while System Integrity Protection is engaged`. It may still work on older macOS versions, but it is worth knowing before anyone spends time on it.

Also ineffective in the stale state: reopening the port (it opens in 0.00s and is silent at every baud rate, passively and after a ping), power-cycling the machine, and `IOBluetoothDevice.closeConnection()`/`openConnection()` — the close returns success while `isConnected()` stays true.

I filed #2226 for the sleep-related side of this and opened two PRs: #2225 prevents the system standby that creates the state during a roast, and #2228 recognizes an unresponsive port and tells the user to re-pair instead of silently retrying forever. Neither repairs a stale port — that does not appear to be possible from inside the application.

[3 more replies]