artisan-roaster-scope/artisan #1711: Modbus connection in Continious Build Artisan 3.0.3 (aa713e7) don´t work.

2571113, 2024-10-10:
### Problem Description

In today's current continuous build Artisan 3.0.3 (aa713e7) the Modbus connection is not working.
In the continuous build Artisan 3.0.3 (427e565) it works.
So, what ist different? Nothing changes in my setup or configuratuons.

[image]

[artisan.txt](https://github.com/user-attachments/files/17324470/artisan.txt)

### Artisan Version

3.0.3 Continuous Build

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

DIY Roaster Arduino Mega

### Files

- [X] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [X] Screenshots are attached.

- [ ] No files are attached.

MAKOMO, 2024-10-10:
Another pymodbus update brought more fun. Should be fixed in next build. Thanks for testing and reporting!

closed 2024-10-10 by MAKOMO

2571113, 2024-10-10:
Now in Artisan 3.0.3 (fef8ba3) the Modbus Connection works well. The
connection ist faster then ever before, even with a delay of 1.3
Thanks!