artisan-roaster-scope/artisan #1718: Autosave "PDF Report" saves PNG image

l2jc, 2024-10-16:
### Problem Description

Using autosave to save roast logs and PDF Reports results in the logs being saved but the PDF Report being a PNG image of the the roast chart. Using just "PDF" saves a PDF for the image but "PDF Report" which contains more information of the roast log doesn't appear to work.

### Artisan Version

3.0.0

### Computer OS and Version

Windows 11

### Connected devices or roasting machine

Mill City Roaster 500g

### Files

- [ ] A sample profile *.alog file, renamed to *.alog.txt is attached.

- [ ] A settings file (Help>> Save Settings) *.aset, renamed to *.aset.txt is attached.

- [ ] Screenshots are attached.

- [X] No files are attached.

roasterdave, 2024-10-16:
I can not reproduce this here on Windows 11 using Artisan v3.0.0 or later.  

Please provide a sample profile .alog file where this happened and your .aset settings file (Help>> Save Settings).  Rename each file to end in "**.txt**" then drag-drop the files onto a reply box in this thread here on GitHub.  Do not send in an email reply.

2571113, 2024-10-17:
I have been using all versions of artisan V3.0.0 up to the current continuous build Artisan 3.0.3 (899462a) with autosave for the last 6 weeks. (Windows 11)

I checked the folder with the autosaved PDFs. There are only PDF files there. Maybe a reset to factory settings and reloading the settings file will help.

l2jc, 2024-10-17:
Sample alog and aset files attached. 

I've just tried a factory reset and reloading settings and it still produces the same issue. To clarify, autosaving "PDF" works fine. It is when autosaving "PDF Reports" that it saves a PNG of the roast chart...no report file to be found.

[1760 Ethiopia Yirgacheffe Halo Beriti 24-10-16.txt](https://github.com/user-attachments/files/17417770/1760.Ethiopia.Yirgacheffe.Halo.Beriti.24-10-16.txt)
[artisan-settings 20241017.txt](https://github.com/user-attachments/files/17417771/artisan-settings.20241017.txt)

beanoccio, 2024-10-17:
Ok, autosaving "PDF Reports" was not used in my earlier Roasts with v3.0.0. (only autosaving "PDF")
I have now tested autosaving "PDF Reports" in the current Artisan 3.0.3 (899462a). Everything seems to be working fine (again).
A PDF with all the details is saved. Identical to how it works manually via the menu.

roasterdave, 2024-10-17:
@l2jc Thank you for the additional information.  I still can not reproduce this with Artisan 3.0.0 and did confirm we are both running the same build.

Inspecting the code there exists a check for a necessary library needed specifically to write the PDF Roast Report.  If the library is missing the PNG file will be written.  I have verified that the Artisan v3.0.0 release installs this library and is thus able to write the PDF Roast report.  Your installation may be corrupt somehow.  I recommend reinstalling Artisan.  

Before reinstalling, it will help me to try to diagnose what happened to your installation if you would do the following.  
- Open a command window (click the Windows key and type `cmd` followed by Enter)
- At the prompt enter:  `dir /s "\program files\artisan\_internal" > dirlist.txt`
- Optionally zip that file and dag it onto a reply in this issue.
- You should then try reinstalling Artisan and check if the problem is resolved.

Thanks for your support.

Terracotta-6, 2024-10-18:
Artisan v3.0.3 (56b381d, 0)  During the function test, we inspected multiple output formats. Autosave performs normally in PDF, PDF Report, SVG, PNG, JPEG, CSV and JSON formats and can be output smoothly. However, it is currently found in the test that this function cannot output in BMP format. In addition, file list files of the Win10 system.
[dirlist.txt](https://github.com/user-attachments/files/17434524/dirlist.txt)

MAKOMO, 2024-10-18:
Support for the image format BMP has been removed

l2jc, 2024-10-21:
@roasterdave Here is the dirlist.

[dirlist.txt](https://github.com/user-attachments/files/17467066/dirlist.txt)

roasterdave, 2024-10-22:
Thank you for providing this additional information.  There are no files missing from the installation.  That makes this all the more puzzling.  If you look at the menu File>> Report>> Roast is the PDF choice active or is it grayed out?  

At this point I recommend installing the [continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).  If the problem persists we will keep working on it.

l2jc, 2024-10-22:
The File > Report > Roast > PDF is available. That is what I've been using (manually) this whole time. It was only recently that I saw there was an autosave function for it too that I wanted to try it.

MAKOMO, 2024-10-23:
We finally could reproduce the issue (it happens not on autosave at the end of the roast, but on pressing Save from the File menu manually) and fix it. The fix is contained in the just being produced [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).

Thanks for reporting!